espruino/Espruino #649: Add I2C master support to esp8266

tve, 2015-10-12:
The initial proposal is to use a slightly modified version of the Espressif i2c library (there is no i2c hardware in the esp8266).

tve, 2015-10-12:
Q: What is `jshI2CSetup` supposed to do if the parameters passed to it are invalid/impossible? For example, if it's asked to be a slave and it can't do that?

gfwilliams, 2015-10-12:
I think it just calls jsError - generally it outputs a list of pins that would be allowed for that peripheral (I'd see the STM32 implementation). Swapping to return true/false too would be cool though.

Do you think it'd be possible to actually implement Software I2C ourselves? It's already done for SPI.

There's a bug for it here: https://github.com/espruino/Espruino/issues/549

It looks pretty trivial to make our own implementation - looking at the Arduino one anyway.

That way it'd add functionality to normal Espruino boards as well.

tve, 2015-10-13:
My current plan is to slap in the esp8266 specific sw i2c. I'd like to see that working first. But I agree that the right way would be a hardware independent implementation using gpio and perhaps timers. I need to read up on how all that hardware timer stuff works in espruino first...

gfwilliams, 2015-10-13:
With I2C, timing isn't generally a big issue (as far as I can sell) so it'd really be a matter of just bashing it out as fast as possible.

That link above mentioned Serial, and timers would be handy there - especially on the receive side of things.

tve, 2015-10-17:
This is implemented. Please reopen if there are issues.

closed 2015-10-17 by tve