espruino/Espruino #653: Add an un-supported board for ESP8266

nkolban, 2015-10-13:
In the boards page found here:

http://www.espruino.com/Reference#boards

It is approaching time to consider adding the ESP8266.  I suggest a picture of the NodeMCU board with content for NodeMCU and commentary that any ESP8266 hosting board will work (ESP-1, ESP-12, Huzzah and more).  Since this would appear on a primary Espruino page, permission and concurrence from @gfwilliams must be obtained.

gfwilliams, 2015-10-13:
Yes, that'd be fine.

You'll want to make sure the image and CSS in BOARD.py matches up nicely first.

IIRC `scripts/build_board_docs.py` is the thing that builds the HTML you need to check

nkolban, 2015-10-16:
A somewhat related forum discussion can be found here:

http://forum.espruino.com/conversations/276005/

tve, 2015-10-17:
#672 provides 90% of what's needed for an esp8266_12 board. Someone else will have to do an esp8266_01 board :-)

nkolban, 2015-10-17:
What is the background to having multiple ESP8266 boards as opposed to just one?  I would have assumed that we would have had one "logical" ESP8266 board encompasing ESP-1, ESP-12, NodeMCU devkit, Hazzah and others.

tve, 2015-10-17:
In the Makefile we need several targets in order to distinguish the various amounts of flash and the completely different linking process for the larger flash sizes with OTA update capability.
I think it also makes sense to have two boards, one for esp-01 and one for esp-12 because the available pins are very different and the pinout is also different.
Finally the flash save area address is different between the non-OTA and the OTA versions because it needs to fit into some free space.
If you manage to combine them, great :-)

nkolban, 2015-10-18:
I don't have any great ideas for the Makefile/build system ... maybe we can all put our heads to it as a community.  If it weren't for the build system, would you be in favor of a single ESP8266 logical board with flash memory variants? ... Eg. ESP8266 512K, ESP8266 1MB, ESP8266 4MB ... etc?

tve, 2015-10-18:
Honestly, I don't care too much. The esp-01 is dead for me anyways... But at this point I'm not volunteering to refactor...

tve, 2015-12-07:
esp8266 is now added to web pages :-)

closed 2015-12-07 by tve