espruino/Espruino #654: ESP8266 save() doesn't work?

rakeshpai, 2015-10-13:
It seems that `save()` from the WebIDE doesn't work. When I do a `save()`, it appears to work successfully:
```
Erasing Flash.....
Writing...
Compressed 12276 bytes to 643
Checking...
Done!
```
A subsequent `dump()` displays the code as well, so everything looks good so far.

However, once I `reset()`, the saved code isn't executed. A look at `dump()` returns nothing, implying that there's no code to be loaded. Somehow, the code seems to be getting lost on a `reset()`.

gfwilliams, 2015-10-13:
What if you power down and back up?

`reset()` should reset the device *without* loading code. `load()` is the one you want if you want to go back to the code you saved.

rakeshpai, 2015-10-13:
Ah. Sorry. My bad. Works fine after a physical reset / power cycle.

closed 2015-10-13 by rakeshpai