espruino/Espruino #660: Allow for pinMode input_pullup in ESP8266

rakeshpai, 2015-10-14:
It appears that `input_pullup` isn't currently implemented. I'm using the following code:
```
var button = new Pin(14);

pinMode(button, "input_pullup");

setWatch(function() {
	print("Button clicked");
}, button, {repeat: true});
```

I have a button connected between GPIO14 and ground. When I press the button the first time, the `print` gets called correctly. However, subsequent button presses don't print anything.

If I then change the `pinMode` to `input`, and add a 10k pull up resistor between GPIO14 and 3.3v, the print gets called as expected upon every button press.

My guess is that that in the first case, the internal pullup resistors in the esp8266 aren't actually enabled, so the pin floats near ground after the first button press, never rising again.

nkolban, 2015-10-14:
There appears to be ESP8266 SDK functions called `PIN_PULLUP_EN` and `PIN_PULLUP_DIS`.  Not yet sure if those are leveraged in the Espruino code base yet, but it seems that we have a pull-up capability in ESP8266 which is a good start.  Not seeing a "pull-down" ability though.

opichals, 2015-10-14:
https://github.com/nekromant/esp8266-frankenstein/blob/master/src/cmd_ds18b20.c#L176

rakeshpai, 2015-10-14:
Just a heads-up - @tve mentioned something in Gitter today about this already implemented in his fork. We should check with him before duplicating effort.

nkolban, 2015-10-14:
I found the source of the error.  The Espruino code does indeed honor the pull-up ability ... however, in the implementation of `jshPinWatch` for ESP8266 we currently set the state of the pin to be `input` before we register an interrupt handler for it.  The thinking here is that there is no point in registering an interrupt handler for an `output` pin.  However, the flag for setting it as input disregards the concept of `input with pullup` and hence we force it to be input only.   The design change I propose is that when jshPinWatch is called it first checks to see if the pin is in any input mode ... if it is, leave it alone.  If it isn't set it to input (simple).  That way, a JS programmer can pre configure the pin to be input or input with pull-up and whatever its current state is when jshPinWatch is caller, will still be honored.

gfwilliams, 2015-10-14:
@nkolban there's something called `jshGetPinStateIsManual` (or similar) which is used in the STM32 implementation. It sets the pin to an input if the state wasn't set, but if it was it leaves it as-is.

nkolban, 2015-10-14:
WOW!!!   Perfect ... exactly what I needed.

nkolban, 2015-10-15:
Build uploaded.  Awaiting @rakeshpai validation.

rakeshpai, 2015-10-15:
Verified. Works perfectly! You're awesome, @nkolban! Thanks!

closed 2015-10-15 by rakeshpai