espruino/Espruino #661: Provide a way to bypass the automatic load() on ESP8266

rakeshpai, 2015-10-14:
I've "bricked" my esp-12e. I uploaded some code that is causing the chip to reset due to the watchdog timer. This is what I see over serial:
```
wdt reset
load 0x40100000, len 29488, room 16 
tail 0
chksum 0xfa
load 0x3ffe8000, len 2108, room 8 
tail 4
chksum 0x08
load 0x3ffe8840, len 28828, room 4 
tail 8
chksum 0x09
csum 0x09
�n��sys=68168 rtc=14974437��;l�Loading 3438 bytes from flash...
Got here

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x40100000, len 29488, room 16 
tail 0
chksum 0xfa
load 0x3ffe8000, len 2108, room 8 
tail 4
chksum 0x08
load 0x3ffe8840, len 28828, room 4 
tail 8
chksum 0x09
csum 0x09
�n��sys=68170 rtc=15116921��;l�Loading 3438 bytes from flash...
Got here
```
The `Got here` message is from my code that's running on the chip, so it seems that the espruino interpretter runs correctly, and the loading of the code works fine too. It's probably something wrong with my code that causes it to mess up.

However, now that the chip is stuck, I can't seem to do anything with it to fix it. I've tried reflashing the chip, but that didn't seem to erase the code that was uploaded. Even after the reflash, my code is still picked up and executed, causing the reset again. Since the chip isn't responding, I can't `reset()` over serial to prevent the code from loading.

The [troubleshooting](http://www.espruino.com/Troubleshooting) page has hardware solutions to this problem, but there's understandably nothing there for the esp8266 chips.

Is this something we can build? Meanwhile, could you suggest something even as a stop-gap so that I can "unbrick" my chip.

gfwilliams, 2015-10-14:
@nkolban `jsiInit` takes a boolean - which should be false if `BTN1` (or similar) is pressed - it'll stop it auto-loading code.

nkolban, 2015-10-14:
@rakeshpai  The JS program that was previously saved is stored as bytes in flash memory.  @tve added that area and I don't know where within the flash memory it is saved.  We have to study and ask @tve where the code is saved.  Once we know where it is saved, we can then use the firmware upload toold to write binary zeros at that location and the previously saved program will be gone.

@gfwilliams Thanks for that ... unfortunately, an ESP8266 doesn't have any "buttons" by default.  We are going to have to give some thought to how we can signal Espruino not to load at boot.  I suspect what we will do is make a configuration parameter that identifies a GPIO and tests that at startup.

gfwilliams, 2015-10-14:
... other option is to wait ~1 sec after power on, and if someone's been bashing Ctrl+C then you know not to load stuff when you call jsiInit

rakeshpai, 2015-10-14:
If we are to go with the external GPIO method, we might really only have GPIO2 to work with, if we want to support the ESP-01.
[image]

I'm not sure how the ctrl+c method would work, but if we could come up with a method that's independent of the WebIDE, I'd prefer that.

nkolban, 2015-10-14:
Unfortunately, I don't think EITHER GPIO0 or GPIO2 can be used as signals at boot time.  The GPIOs known as GPIO0, GPIO2 and GPIO15 are all examined by ESP8266 itself at boot time to determine what the ESP8266 itself should do.  For example GPIO0 = low means flash while high means run.  GPIO2 has similar semantics.  After boot, they can be used as GPIO but in this story, it won't help us.

tve, 2015-10-14:
Flash blank.bin to 0x7C000. The flash layout is documented in https://github.com/espruino/Espruino/wiki/ESP8266-Design-Notes

nkolban, 2015-10-15:
The `esptool` found [here](https://github.com/themadinventor/esptool) has an option called `erase_flash` that can erase all the flash of the ESP effectively resetting it.  A request for an erase option for `esptool-ck` has been [submitted](https://github.com/igrr/esptool-ck/issues/15).

MaBecker, 2016-12-09:
@rakeshpai can this be closed ?

closed 2016-12-10 by nkolban