espruino/Espruino #662: Take the ESP8266 board support out of beta

nkolban, 2015-10-14:
This work item is to track the activities necessary to be performed to take the ESP8266 board support out of beta into what ever state we consider it to be when no longer beta.

For example, currently we have debugging on by default, we need to switch that to off when the board "ships".

(I'm using loose words here like "ships" to mean that it no longer is in beta ... if there is some desired phraseology we want to use, I'm happy to follow along).

@gfwilliams From an Espruino perspective, when development off the ESP8266 board port reaches a conclusion and we communally decide it is ready for end user consumption, what should we call that phase?  I'm presuming you are not keen in calling it "supported" since that is a phrase that has been reserved for the purchasable boards?

gfwilliams, 2015-10-14:
To be honest I'm not sure - I'm very happy to say 'out of beta'?

gfwilliams, 2015-10-14:
Just to add, from my point of view we need:

* BOARD.py to generate an HTML page with an image and pin names that line up.
* Potentially at the very least a BOARD.py for ESP01 and ESP12
* Changing `ESP8266WiFi` to a module, and bringing the method names more in line with the rest of Espruino

ebirger, 2015-10-14:
Perhaps a purchasable esp8266 Espruino board? :)

nkolban, 2015-10-14:
@ebirger I like the notion ... but what would it be?  All I could imagine is something like a NodeMCU DevKit pre-flashed with Espruino.

gfwilliams, 2015-10-14:
Yes, it'd be a good plan - especially if @tve's OTA upgrade would allow firmware updates with the Web IDE (that's probably needed for selling a 'proper' board - every day users will be less happy with random Python scripts for updates :) ). 

I might be able to have a word with Adafruit and Seeed about it - I'd like to get in there before a loads of clones start appearing on Aliexpress/eBay/etc :)

gfwilliams, 2015-10-14:
Oh - another thing that'd definitely be nice for getting out of beta is the context switching - allowing debug, and not having long bits of code reboot the device would be properly handy

tve, 2015-10-14:
I would recommend tagging the issues that need to be resolved to get out of beta. IMHO getting SPI, and I2C finished and tested is a pre-req. As well as a telnet terminal.

tve, 2015-10-15:
Something else that I think is needed is for Gordon to do a full code review of the esp8266 specific code. Not to check on the esp8266 implementation per-se but to make sure we're not overlooking Espruino integration requirements or optimizations. I've been browsing quite a bit of Espruino core code, but I'm still far from knowledgeable, so it's quite likely that I missed stuff or make mistakes where the core expects specific semantics but I implemented something slightly different.

gfwilliams, 2015-10-15:
Yes, probably a good idea - I've tried to take a glance now and then, but at the moment I have virtually no free time, so I'm afraid it'd have to wait a while.

MaBecker, 2016-03-01:
what do think gordon and tve, is now time ?

gfwilliams, 2016-03-01:
I think this bug can just be closed? I removed the beta text before I did the release on Espruino.com I think.

closed 2016-03-01 by nkolban