espruino/Espruino #665: Add support for Serial access for ESP8266 board

nkolban, 2015-10-15:
Serial processing (UART0 and UART1) has not been tested and may not work.  This issue looks to see what might be needed to be done to support Serial access from JavaScript through the ESP8266 serial interfaces.

tve, 2015-12-07:
From Gordon:
> @tve it'd need to implement jshUSARTSetup - I don't know if that's actually done right now?
> You can move the console out of the way with something like LoopbackA.setConsole() and then you can use Serial1 for whetever you want...

http://forum.espruino.com/conversations/278532/#comment12666182

SpenceKonde, 2016-02-17:
Any updates on this? 
Getting a working serial implentation on ESP8266 would be a real boon.

gfwilliams, 2016-02-18:
Have got Serial1.setup working with https://github.com/espruino/Espruino/commit/cb0232b70e054133c3e3babe19dc675529b5a372

So you can read and write Serial1 when you move the console elsewhere. I'm less sure whether Serial2 works yet

MaBecker, 2017-07-23:
Serial1
```
// disable console on Serial1
Serial1.setConsole(false)
// now you can use Serial1 for other stuff
```

Serial2 can only be used to transmit data over GPIO2.

```
// disable debug to free GPIO2
require("ESP8266").setLog(0)
// now you can use Serial2-TXD for other stuff 
Serial2.print("Hello");
```
So lets close this.

closed 2017-07-23 by MaBecker

gfwilliams, 2017-07-24:
Thanks! Just so you know, `Serial1.setConsole(false)` won't disable the console on Serial1 - it just allows it to be hot-swapped if (for example) Telnet connects.

To move it out the way you need to put it somewhere else with (for example) `Serial2.setConsole(false)` or `LoopbackA.setConsole(false)`