espruino/Espruino #668: esp8266 reset

HyGy, 2015-10-16:
Hi,

There is an example code, what is reset some time of running and changing the state of the led with the provided link. It is more stable than with yesterday (2015-10-15) binaries, but it crashez after a time.

```javascript
/*
var GPIO2 = new Pin(2);
var toggle=1;

setInterval(function() {
toggle=!toggle;
  digitalWrite(GPIO2, toggle);
}, 500);
*/

var connectToWifi= function()
{
    var ssid="ssid";
    var pass="pass";

    print('connecting to wifi');

    ESP8266WiFi.init();
    ESP8266WiFi.connect(
        ssid,
        pass,
        function()
        {
            print('connected to wifi');
            var ipInfo = ESP8266WiFi.getIPInfo();
            print("Current IP address is: " + ESP8266WiFi.getAddressAsString(ipInfo.ip)+':'+ ESP8266WiFi.getAddressAsString(ipInfo.netmask));

            print("starting webserver");
            startWebServer();
        }
    );

};

var startWebServer = function()
{

    var GPIO2 = new Pin(2);
    var http = require("http");

    var httpSrv = http.createServer(function(request, response) {
        print(process.memory());

        response.write("<html><body>");
        if (request.url == "/on") {
            response.write("<b>LED os on</b>"); 
            digitalWrite(GPIO2, 0);
            print ("led on");
        } else if (request.url == "/off") {
            response.write("<b>LED os off</b>"); 
            print ("led off"); 
            digitalWrite(GPIO2, 1);

        } else {
            response.write("Welcome to ESP with JS");
        }
        response.end("<br><a href='/on'>Led ON</a><br><a href='/off'>Led OFF</a></body></html>");
    });

    httpSrv.listen(80);

};

connectToWifi();
```

error from the esp8266 board:
```javascript
{ "free": 458, "usage": 565, "total": 1023, "history": 202 }
{ "free": 445, "usage": 578, "total": 1023, "history": 202 }
led off
{ "free": 458, "usage": 565, "total": 1023, "history": 202 }
{ "free": 445, "usage": 578, "total": 1023, "history": 202 }
led on
{ "free": 402, "usage": 621, "total": 1023, "history": 202 }
{ "free": 357, "usage": 666, "total": 1023, "history": 202 }
led off
ERROR: Socket error -1 while sending
{ "free": 458, "usage": 565, "total": 1023, "history": 202 }
>
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)
load 0x40100000, len 30648, room 16
tail 8
chksum 0x4f
load 0x3ffe8000, len 2108, room 0
tail 12
chksum 0x00
ho 0 tail 12 room 4
load 0x3ffe8840, len 28804, room 12
tail 8
chksum 0xb6
csum 0xb6
oîsys=67978 rtc=13955987
don't use rtc mem data
{l{dâÄcdÄùLoading 864 bytes from flash...
> 
```

nkolban, 2015-10-16:
Does it crash while being left along or is there some sequence of browser originated requests that cause it to crash?

HyGy, 2015-10-17:
I need to press the on/off button on the web interface about 20-30 times
rapidly.

(yesterday with the 15.th rom it is crashed some times with out doing
anything)
2015. okt. 17. de. 12:38 ezt írta ("Neil Kolban" <notifications@github.com

markusgritsch, 2015-10-19:
@HyGy can you please try again using version "Build for 2015-10-18"?  I get much better results since #673 got fixed!

HyGy, 2015-10-19:
I got this after clicking fast to off and on:

First try:

```javascript
=undefined
{ "free": 477, "usage": 546, "total": 1023, "history": 242 }
led on
{ "free": 477, "usage": 546, "total": 1023, "history": 242 }
led on
{ "free": 463, "usage": 560, "total": 1023, "history": 242 }
led off
{ "free": 477, "usage": 546, "total": 1023, "history": 242 }
led on
{ "free": 413, "usage": 610, "total": 1023, "history": 242 }
led on
{ "free": 453, "usage": 570, "total": 1023, "history": 242 }
led on
{ "free": 399, "usage": 624, "total": 1023, "history": 242 }
led on
{ "free": 463, "usage": 560, "total": 1023, "history": 242 }
led on
{ "free": 413, "usage": 610, "total": 1023, "history": 242 }
led on
{ "free": 477, "usage": 546, "total": 1023, "history": 242 }
led on
{ "free": 477, "usage": 546, "total": 1023, "history": 242 }
led off
{ "free": 463, "usage": 560, "total": 1023, "history": 242 }
led off
{ "free": 413, "usage": 610, "total": 1023, "history": 242 }
led off
{ "free": 477, "usage": 546, "total": 1023, "history": 242 }
led off
>
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)
load 0x40100000, len 30936, room 16
tail 8
chksum 0x22
load 0x3ffe8000, len 2108, room 0
tail 12
chksum 0x81
ho 0 tail 12 room 4
load 0x3ffe8840, len 28864, room 12
tail 4
chksum 0xf5
csum 0xf5
don't use rtc mem data
Disconnected
```

second try:

```javascript
{ "free": 209, "usage": 814, "total": 1023, "history": 28 }
led on
{ "free": 193, "usage": 830, "total": 1023, "history": 28 }
led off
{ "free": 171, "usage": 852, "total": 1023, "history": 28 }
led off
{ "free": 249, "usage": 774, "total": 1023, "history": 28 }
led off
{ "free": 269, "usage": 754, "total": 1023, "history": 30 }
led off
{ "free": 269, "usage": 754, "total": 1023, "history": 30 }
led on
ERROR: Socket error -1 while sending
{ "free": 270, "usage": 753, "total": 1023, "history": 30 }
led on
ERROR: Socket error -1 while sending
{ "free": 270, "usage": 753, "total": 1023, "history": 30 }
led on
ERROR: Socket error -1 while sending
>
Disconnected
> 
```

HyGy, 2015-10-19:
I created another automatic test case for the same problem yesterday.

Just need to replace this part of the code:

```javascript
        } else {
            response.write("Welcome to ESP with JS");
        }

        response.write("<script>if (document.location.pathname=='/on') { document.location.href='/off' } else { document.location.href='/on' }</script>");
        response.write("<br><a href='/on'>Led ON</a><br><a href='/off'>Led OFF</a>");
        response.end("</body></html>");
    });

    httpSrv.listen(80);
```

It's running now. I think this automatic "test" is working better until now. (I just get some socket errors, and one disconnect from the espruino web ide) But I let it run more.

But the original problem clicking with the mouse is still occours, as I copied to my post before.

HyGy, 2015-10-20:
I let it run and I got this:

```javascript
led on
{ "free": 482, "usage": 541, "total": 1023, "history": 231 }
led off
{ "free": 482, "usage": 541, "total": 1023, "history": 231 }
led on
{ "free": 482, "usage": 541, "total": 1023, "history": 231 }
led off
>
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)
load 0x40100000, len 30936, room 16
tail 8
chksum 0x22
load 0x3ffe8000, len 2108, room 0
tail 12
chksum 0x81
ho 0 tail 12 room 4
load 0x3ffe8840, len 28864, room 12
tail 4
chksum 0xf5
csum 0xf5
don't use rtc mem data
;l{dãÄcdÄùLoading 5279 bytes from flash...
connecting to wifi
connected to wifi
Current IP address is: 192.168.1.67:255.255.255.0
starting webserver
> 
```

nkolban, 2015-11-17:
Diagnosis was performed and a culprit was believed to have been found.  On a network request (HTTP GET) that failed to get a response from the partner, the ESP8266 port was NOT releasing resources.  See issue #595.  Because of this, we would leak resources over prolonged time.   That particular issue has been corrected.  We are now at a point where we believe a retest is warranted as we believe that will have cured up this problem.

HyGy, 2015-11-18:
okay, I have time to upload the latest firmware, with the same code without any onewire device, it just create a random number, and send it to thingspeak. Here is what I got:

>process.env
={
  "VERSION": "1v81.778",
  "BUILD_DATE": "Nov 13 2015",
  "BUILD_TIME": "22:27:56",
  "GIT_COMMIT": "60e92f61065e2a3f5998dfec73018469ed3d6728",

```javascript
get callback!
connection getdata:
{ "free": 107, "usage": 916, "total": 1023, "history": 9 }
ERROR: Out of Memory!
WARNING: Unable to create string as not enough memory
WARNING: Unable to create string as not enough memory
WARNING: Out of memory while appending to array
WARNING: Unable to create string as not enough memory
WARNING: Unable to create string as not enough memory
WARNING: Out of memory while appending to array
```

I tried to turn on debuging, but I got minimal information for the debug console:

```javascript
Restart info:
  reason:   6=reset pin
  exccause: 0
  epc1:     0
  epc2:     0
  epc3:     0
  excvaddr: 0
  depc:     0
Flash map 512KB:256/256, manuf 0xE0 chip 0x4016
Heap: 5056
Variables: 1023 @12ea = 12276bytes
Time sys=323398 rtc=60533
mode : sta(18:fe:34:fa:3f:80)
add if0
f r0, > initDone
> jshInit
RTC read: 2115955144 262570869 0xbd53886a (0x37ed9a82)
RTC: cannot restore time
UStimer init
< jshInit
> jswrap_ESP8266_init
> jswrap_ESP8266WiFi_init
Wifi init, mode=1
< jswrap_ESP8266WiFi_init
< jswrap_ESP8266_init
> ESP8266: jshPinSetState 0, GPIO_OUT, pup=0, od=0
> ESP8266: jshPinSe
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
```

nkolban, 2015-11-18:
Comment out the use of GPIO2 in your code ... GPIO2 is also the pin used for debug TX to UART1.  Re-run your test and let us capture the debug a second time.

HyGy, 2015-11-18:
Can I send a debug message from the js console to the other?

nkolban, 2015-11-18:
Sorry ... I don't understand that.   Built into the Espruino port is logic that will write debug to the debug UART of an ESP8266.  That is hard-wired to UART1 which is GPIO 2 on an ESP8266 device.

HyGy, 2015-11-18:
Yes, and I want to know, how can I trigger this debug from the "simple espruino web js" console. To test is it working or not while the device is running.

tve, 2015-12-07:
Would you mind checking this again with the new build that has the updated Wifi and socket code? https://s3.voneicken.com/espruino/espruino_1v83.tve_fix-sockets_e62bff7_esp8266.tgz

tve, 2015-12-18:
Please reopen if this problem recurs.

closed 2015-12-18 by tve