espruino/Espruino #673: ESP8266 resets on fast consecutive web requests

markusgritsch, 2015-10-17:
I am using version 2015-10-17 with the following simple web server:

```javascript
ESP8266WiFi.beAccessPoint('ESP_1710')
ESP8266WiFi.init();

var GPIO2 = new Pin(2);
var http = require("http");

var httpSrv = http.createServer(function(request, response) {
  print(process.memory());
  print(ESP8266WiFi.getState());

  response.write("<html><body>");
  if (request.url == "/on") {
    response.write("<b>LED os on</b>");
    digitalWrite(GPIO2, true);
    print ("led on");
  } else if (request.url == "/off") {
    response.write("<b>LED os off</b>");
    print ("led off");
    digitalWrite(GPIO2, false);
  } else {
    response.write("Welcome to ESP with JS");
  }
  response.end("<br><a href='/on'>Led ON</a><br><a href='/off'>Led OFF</a></body></html>");
});

httpSrv.listen(80);
```

It works fine, but when performing two or more fast consecutive web requests (double clicking on the "Led ON"-link), the ESP8266 resets with the following ASSERTION

```
ASSERT(pSocketData->currentTx == ((void *)0)) FAILED AT libs/network/esp8266/network_esp8266.c:945
  #1[r1,l2] Object {
    #2[r1,l2] Name String [1 blocks] "ÿ"      #3[r1,l2] Object {
        #6[r1,l2] Name String [2 blocks] "timers"          #8[r2,l1] Array(0) [ ]
        #9[r1,l2] Name String [2 blocks] "watches"          #11[r2,l1] Array(0) [ ]
        #14[r1,l2] Name String [1 blocks] "net"          #38[r1,l3] String [2 blocks] "\x04\x00\x00\x00\x00ÿÿÿ"
        #20[r1,l2] Name String [2 blocks] "history"          #29[r1,l1] Array(43) [
            #28[r1,l2] Name Integer 0              #19[r1,l1] String [6 blocks] "ESP8266WiFi.beAccessPoint('ESP_1710')"
            #44[r1,l2] Name Integer 8              #41[r1,l1] String [3 blocks] "ESP8266WiFi.init();"
            #61[r1,l2] Name Integer 10              #40[r1,l1] String [6 blocks] "var ipInfo = ESP8266WiFi.getIPInfo();
            #72[r1,l2] Name Integer 11              #73[r1,l1] String [11 blocks] "print(\"Current IP address is: \" +
            #68[r1,l2] Name Integer 12              #47[r1,l1] String [4 blocks] "var GPIO2 = new Pin(2);"
            #90[r1,l2] Name Integer 13              #93[r1,l1] String [4 blocks] "var http = require(\"http\");"
            #170[r1,l2] Name Integer 14              #173[r1,l1] String [72 blocks] "var httpSrv = http.createServer(fu
            #168[r1,l2] Name Integer 15              #260[r1,l1] String [3 blocks] "httpSrv.listen(80);"
            #36[r1,l2] Name Integer 23              #280[r1,l1] String [3 blocks] "process.memory()"
            #59[r1,l2] Name Integer 42              #418[r1,l1] String [4 blocks] "ESP8266WiFi.getState()"
          ]
        #97[r1,l2] Name String [2 blocks] "modules"          #99[r1,l1] Object {
            #89[r1,l2] Name String [1 blocks] "http"              #100[r2,l1] ...

          }
        #265[r1,l2] Name String [2 blocks] "HttpS"          #267[r1,l1] Array(1) [
            #268[r1,l2] Name Integer 0              #254[r3,l1] ...

          ]
        #297[r1,l2] Name String [2 blocks] "HttpSC"          #300[r1,l2] Array(21) [
            #327[r1,l3] Name Integer 20              #413[r1,l2] Object {
                #405[r1,l2] Name String [2 blocks] "__proto__"                  #264[r2,l1] ...

                #334[r1,l2] Name String [1 blocks] "type"= int 1

                #344[r1,l2] Name String [1 blocks] "res"                  #351[r1,l2] Object {
                    #383[r1,l2] Name String [2 blocks] "__proto__"                      #281[r2,l1] ...

                    #278[r1,l2] Name String [1 blocks] "code"= int 200

                    #350[r1,l2] Name String [1 blocks] "hdr"                      undefined
                    #338[r1,l2] Name String [1 blocks] "dSnd"                      #372[r1,l2] String [19 blocks] "HTTP
                    #404[r1,l2] Name String [2 blocks] "close"                      #389[r1,l1] Bool true
                  }
                #366[r1,l2] Name String [1 blocks] "svr"                  #254[r3,l1] ...

                #323[r1,l2] Name String [1 blocks] "sckt"= int 22

                #320[r1,l2] Name String [1 blocks] "dRcv"                  #373[r1,l1] String [1 blocks] ""
                #414[r1,l2] Name String [2 blocks] "headers"                  #322[r1,l1] Object {
                    #362[r1,l2] Name String [1 blocks] "Host"                      #377[r1,l1] String [2 blocks] "192.1
                    #382[r1,l2] Name String [2 blocks] "User-Agent"                      #378[r1,l1] String [10 blocks]
                    #289[r1,l2] Name String [2 blocks] "Accept"                      #394[r1,l1] String [9 blocks] "tex
                    #374[r1,l2] Name String [3 blocks] "Accept-Language"                      #402[r1,l1] String [4 blo
                    #379[r1,l2] Name String [3 blocks] "Accept-Encoding"                      #359[r1,l1] String [3 blo
                    #410[r1,l2] Name String [2 blocks] "Referer"                      #358[r1,l1] String [4 blocks] "ht
                    #367[r1,l2] Name String [2 blocks] "Connection"                      #403[r1,l1] String [2 blocks]
                  }
                #398[r1,l2] Name String [2 blocks] "method"                  #354[r1,l1] String [1 blocks] "GET"
                #409[r1,l2] Name String [1 blocks] "url"                  #319[r1,l1] String [1 blocks] "/on"
                #345[r1,l2] Name String [1 blocks] "hdrs"                  #307[r1,l1] Bool true
              }
          ]
      }
    #15[r1,l2] Name String [2 blocks] "EventDetails"      undefined
    #22[r1,l2] Name String [2 blocks] "ESP8266WiFi"      #18[r1,l1] NativeFunction 0x402223ac (1) { }
    #30[r1,l2] Name String [2 blocks] "process"      #31[r1,l1] NativeFunction 0x402224a0 (1) { }
    #42[r1,l2] Name String [2 blocks] "ipInfo"      #55[r1,l1] Object {
        #56[r1,l2] Name String [2 blocks] "__proto__"          undefined
        #62[r1,l2] Name String [1 blocks] "ip"= int 0

        #63[r1,l2] Name String [2 blocks] "netmask"= int 0

        #65[r1,l2] Name String [1 blocks] "gw"= int 0

      }
    #52[r1,l2] Name String [2 blocks] "Restart"      undefined
    #57[r1,l2] Name String [2 blocks] "GPIO2"      #92[r1,l1] Pin 2
    #71[r1,l2] Name String [1 blocks] "Pin"      #67[r2,l1] NativeFunction 0x40234140 (9) {
        #69[r1,l2] Name String [2 blocks] "prototype"          #86[r1,l1] Object {
            #87[r1,l2] Name String [2 blocks] "constructor"              #67[r2,l2] ...

          }
      }
    #91[r1,l2] Name String [1 blocks] "http"      #100[r2,l1] NativeFunction 0x40222420 (0) { }
    #172[r1,l2] Name String [2 blocks] "httpSrv"      #254[r3,l1] Object {
        #255[r1,l2] Name String [2 blocks] "__proto__"          #251[r2,l1] Object {
            #252[r1,l2] Name String [2 blocks] "constructor"              #248[r1,l1] NativeFunction 0x40222410 (1) {
                #249[r1,l2] Name String [2 blocks] "prototype"                  #251[r2,l2] ...

              }
          }
        #258[r1,l2] Name String [1 blocks] "type"= int 1

        #257[r1,l2] Name String [2 blocks] "#onconnect"          #167[r1,l1] Function {
            #166[r1,l2] Name Param "request"               undefined
            #162[r1,l2] Name Param "response"               undefined
            #247[r1,l2] Name String [1 blocks] "ÿcod"              #164[r1,l1] String [64 blocks] "{\nprint(process.mem
          }
        #269[r1,l2] Name String [1 blocks] "port"= int 80

        #270[r1,l2] Name String [1 blocks] "sckt"= int 1

      }
    #279[r1,l2] Name String [2 blocks] "ESP8266State"      undefined
    #285[r1,l2] Name String [2 blocks] "httpSRq"      #169[r2,l1] NativeFunction 0x40222400 (1) {
        #277[r1,l2] Name String [2 blocks] "prototype"          #264[r2,l1] Object {
            #274[r1,l2] Name String [2 blocks] "constructor"              #169[r2,l2] ...

          }
      }
    #43[r1,l2] Name String [2 blocks] "httpSRs"      #33[r2,l1] NativeFunction 0x402223f0 (1) {
        #275[r1,l2] Name String [2 blocks] "prototype"          #281[r2,l1] Object {
            #290[r1,l2] Name String [2 blocks] "constructor"              #33[r2,l2] ...

          }
      }
  }

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 30936, room 16
tail 8
chksum 0x37
load 0x3ffe8000, len 2108, room 0
tail 12
chksum 0x02
ho 0 tail 12 room 4
load 0x3ffe8840, len 28804, room 12
tail 8
chksum 0x5a
csum 0x5a
don't use rtc mem data
```

nkolban, 2015-10-18:
Starting examination.

nkolban, 2015-10-18:
Fix now available in latest builds.  Waiting on @markusgritsch to validate.

markusgritsch, 2015-10-19:
Works flawlessly.  Thanks a lot.  Keep up the good work!

markusgritsch, 2015-10-19:
While two or three fast consecutive clicks can be handled now, I still can crash the ESP8266 by holding CTRL+R in the browser.  I think it is caused by running out of heap, since the output of ESP8266WiFi.getState() indicates this.

nkolban, 2015-10-19:
@markusgritsch I'm going to close this issue if that is ok with you.  An issue  might want to be opened if we have a story about running out memory with requests coming in too fast.  There might not be much we can do about that as each new concurrent socket takes resources that are only released when the socket closes ... and it might be rapid CTRL+Rs are producing more socket requests at a rate faster than they are being detected as closed.   However... that still shouldn't crash the device.

closed 2015-10-19 by nkolban

markusgritsch, 2015-10-19:
It's ok with me.  If it would not crash, even under heavy hammering with requests, it would be even more awesome :)

HyGy, 2015-10-20:
I think this is the same problem as #668 and I think it is not good now.

markusgritsch, 2015-10-20:
@HyGy No, the assertion mentioned in the first comment was caused by something different, which Neil has fixed now. The remaining instability in #668 is maybe because of running out of heap, maybe because of something else, but it does not trigger the above assertion.