espruino/Espruino #675: Add ESP8266 board to the Web IDE module lookup system

nkolban, 2015-10-19:
When we use the Web IDE, modules that are referenced by `require` are showing up as not found.  It is believed that this is caused by the espruino.com web site not having a board definition for the ESP8266 in its configuration.   This issue asks for examination and resolution of that issue to prevent warnings from appearing within the Web IDE when being used with an ESP8266 module.

gfwilliams, 2015-10-20:
Which specific modules? If the built-in ones are creating errors, that'd definitely be the problem - if it's others then it'll be something else.

nkolban, 2015-10-20:
I believe the one reported to me was a `require('http')`.

gfwilliams, 2015-10-21:
Ok, yes - so that'd be it then...

nkolban, 2015-10-24:
With the work being done on #589 and #630 this issue is becoming more pressing as use of ESP8266 board now always needs "unknown" `require` modules.

gfwilliams, 2015-10-26:
Will see what I can do. I'm out again this week though - conference season :)

nkolban, 2015-10-26:
From my perspective, this is relatively low priority as everything is functioning even with these warning messages.  However, I believe that "impression" is as much quality as function and some users are expressing "query" over the messages in Web IDE.

nkolban, 2015-11-05:
User @Ducky is reporting the warnings about modules within the Web IDE.  See:

http://forum.espruino.com/conversations/266886/?offset=125#comment12608682

gfwilliams, 2015-11-05:
We now have ESP8266_BOARD and ESP8266_12 boards. Is this going to stay the case? There was some talk on the forum about just having a single binary for both boards?

nkolban, 2015-11-05:
This is where the community should come together and make a decision.  @tve did a lot of work on the OTA flashing and it would be great to involve him for sure.   There is also a work item on a re-design of the Makefile/build system.  I think any significant changes to the story on boards should probably await the decisions on the outcome of a reworked Makefile/build system as changes there would affect the design of boards vs board options.

I am also guessing that if we polled the user community, the majority of users who are reporting the problems are using the firmware images supplied at EspruinoBuilds ... and if that is the case, those are being compiled today using "ESP8266_BOARD".  If there is validity in that notion, then accommodating "ESP8266_BOARD" as a known board today (even if not strategically where we might want to go in the future) might satisfy the majority of users sooner than later.

gfwilliams, 2015-11-09:
Just done. Added the docs at http://www.espruino.com/EspruinoESP8266

Boards reporting ESP8266_BOARD or ESP8266_12 will now pull in json from espruino.com, although if we merge stuff into _BOARD at some point then I might kill _12 eventually.

closed 2015-11-09 by gfwilliams