espruino/Espruino #676: New I2C Slave address entry in jshardware.c?

gfwilliams, 2015-10-19:
Hi,

I just noticed the addition of `slaveAddr` in jshardware.h in 606f4adf7b69613fbab3bf64d294d60a282c396a (a cast there had been creating a compile warning :) )

@tve could we just remove this? Looks like it's not implemented in ESP8266, it definitely isn't in STM32 or Linux, and there's no API for calling it - so it's just wasting space at the moment?

The reason I2C slave was never put into Espruino is, like SPI slave, it really requires *guaranteed* quick responses. Sure, it can clock-stretch, but if it's ever over 1ms most things will time out... and you can't really guarantee sub-1ms on Espruino unless you run the JS via an IRQ (which we can't do at the moment).

I think to do it properly we'll have to implement some native code IRQ-based I2C slave implementation (probably faking a bunch of registers) and then expose that to JS via an API... But that's a bigger job and IMO will require a lot more doing than just a `slaveAddr` variable :)

tve, 2015-10-19:
Agreed. And yes, I only made a type change 'cause for some reason 'char' was ending up as unsigned and some function needed to return -1.

gfwilliams, 2015-10-20:
> for some reason 'char' was ending up as unsigned 

Wow - on ESP8266? Hmm, I guess a lot of stuff could do with changing then - that could lead to a few dodgy bugs.

ebirger, 2015-10-20:
Or perhaps add -fsigned-char to the generic CFLAGS

commit bf6ec683 (2015-10-29): remove slaveAddr (fix #676)
closed 2015-10-29 by gfwilliams