espruino/Espruino #679: httpServer on ESP8266 not running after reset and upload source in WebIDE

jumjum123, 2015-10-26:
I'm testing with last version of firmware, running into a lot of strange problems.
Snapshots are only available for this run, there have been so many problems,...
Hope, this is helpful for the experts, its a lot of info, didn't want to miss something.
At the end this seems to be reproducable, at least on my environment.
I've added 3 sets of logs, where the main problem is reproduceable in 3rd log

**Part 1**
- ConnectStatus returns "connecting", for at least 10 minutes
- ConnectStatus returns "wrong password"
- ConnectStatus returns "unknown error"

Have no idea why, but after doing a lot of connecting, reconnecting, reset,
 ... it connected (see Console WebIDE)
*Next started Http-server, opened Firefox and called an url on server.*
Something happened in debug window, nothing else.
Even a console.log in Callback of createServer didn't do anything.
*Next tried a new upload from WebIDE* and got a reset of ESP8266.
Had this behaviour (reset of ESP8266) several times, and usually everything worked fine after that.
Based on a "training" by Neill, there was an exception in phy_check_data_table.
But this is only a part of the problem, and where my knowledge comes to an end. 
Please see my application and all logs. 
Firmware : 1v81.661
Application Url : http://jumspruino.jumware.com/problems/Server_26_Oct_2015/Application.txt
Console WebIDE : http://jumspruino.jumware.com/problems/Server_26_Oct_2015/ConsoleLog.txt
DebugConsole : http://jumspruino.jumware.com/problems/Server_26_Oct_2015/DebugLog.txt

**Part 2**
*Next reconnected to Com4 and uploaded source.*
- ConnectStatus is "unknwon error"

*After stopping WiFi and connecting* again everything looks fine
*Uploaded source again* (don't know why, just did it)
- ConnectStatus is "connected"

*Started httpServer*
*Opened Url from firefox several times*, everything works like crazy, no problems
Console WebIDE : http://jumspruino.jumware.com/problems/Server_26_Oct_2015/ConsoleLog_2.txt
DebugConsole : http://jumspruino.jumware.com/problems/Server_26_Oct_2015/DebugLog_2.txt

**Part 3**
And now the story starts...
*Uploaded source one more time.*
- ConnectStatus is "connected"

*Started httpServer*
*Opened Url from firefox* and after some seconds got a timeout
Console WebIDE : http://jumspruino.jumware.com/problems/Server_26_Oct_2015/ConsoleLog_3.txt
DebugConsole : http://jumspruino.jumware.com/problems/Server_26_Oct_2015/DebugLog_3.txt

From my experience with this problem, I now have to try it again until ESP8266 gets the error as in first logs.
HW-reset doesn't help, ESP8266Wifi.init() does not help, power of does not help, ...

jumjum123, 2015-10-26:
I've added video which shows the reproducable behaviour (at least in my environment)
http://www.jumspruino.jumware.com/problems/Server_26_Oct_2015/video.mp4
After the end of the video, the same story starts again.

nkolban, 2015-10-26:
@jumjum123 Your documentation of problems is absolutely fantastic.  I'm at real-work just now so can't examine the problem just now but please be assured that this is next up when I have Espruino time.

I am pretty sure that I have some guesses ... first is issue #619.  This asks for detailed examination of the Espruino ESP8266 "reset" logic.  Here is what I think is happening in your case.  When you first run, a TCP socket is started in ESP8266 listening on port 80.  This is then bound to the Espruino logic to call back into your own JavaScript.  All works.

Next, you push a new copy of your code to the Espruino/ESP8266 ... and here is where things start to fall down.  Inside Espruino code, the act of pushing a new program is supposed to do a "soft reset".   This means that the state of Espruino running inside the ESP8266 is supposed to be reset to a newly loaded state ... but without doing a hardware restart of the ESP8266.  This is achieved by calling the internal function called `jshReset`.  Ok ... but what does that MEAN?   Well ... what it should do is release all the ACTUAL ESP8266 TCP sockets and also result in Espruino thinking it has no sockets.  What I think is ACTUALLY happening is that Espruino sockets are reset but the socket that was listening on port 80 remains alive at the ESP8266 level.  Since there is now no longer any JavaScript association with that socket, when you connect with the browser, nothing happens.   When you re-load the app multiple times ... it seems to continue to re-load (but not work) for some number of times and then we crash.  I think I can explain this one too.  When you connect with your browser, a new TCP socket connection is formed to the ESP8266 ... and that happens each time you test.  However, the ESP8266 can only maintain a finite number of concurrent sockets (I believe it is 5 by default).  When we reach the limit, it looks like there isn't a "check" made on the returned socket to see if it is good ... and the code merrily continues until BANG ... it tries to access something that isn't properly allocated.

Half the battle in getting things working is having a clue at why they are breaking ... so this theory gives us something to examine and tackle.  In the meantime, if you do an ESP8266 board reset (physically bring RESET pin to low and then high), you should be able to re-re-load.

Apologies for the problems but MANY thanks for the patience and time you spend in building such great reports.

nkolban, 2015-10-26:
Starting examination ...

profra, 2015-10-26:
I am maybe wrong, but... see your source of App...
....  console.log("ConnStatus",ESP8266WiFi.getConnectStatus()); .... ????
You declare ESP8266 not ESP8266WiFi...

jumjum123, 2015-10-26:
I use ESP8266WiFi as written here:https://github.com/espruino/Espruino/blob/master/targets/esp8266/docs/05-User_Guide.md

profra, 2015-10-26:
Yes, thanks, I am a step behind :-)

nkolban, 2015-10-27:
Found a horrible error .... I had misunderstood `espconn_delete`.   I had assumed that this was used against ESP8266 server sockets (which it is) ... however, I had assumed that this was the API to terminate such a socket.  Apparently it isn't.  This is what deletes the memory associated with that socket.

nkolban, 2015-10-27:
Awaiting technical support response on ESP8266 SDK question found here:

http://bbs.espressif.com/viewtopic.php?f=7&t=1290

tve, 2015-12-18:
I believe this is all fixed at this point. Please reopen if not.

closed 2015-12-18 by tve