espruino/Espruino #689: wifi.scan() method leads to a memory overflow on ESP8266 board

yerpj, 2015-10-31:
The scan() method exposed by the 'wifi' module leads to a memory overflow when called 6-7 times successively. The output is 
" ERROR: Out of Memory!
WARNING: Unable to create string as not enough memory
WARNING: Out of memory while appending to array
ASSERT(jsvHasChildren(obj)) FAILED AT src/jsvar.c:2135 " 

This issue was observed on the ESP8266 port of Espruino. I did not try on the Pico with an ESP8266 bounded as a network interface.

nkolban, 2015-10-31:
There is definitely a leak here.  I run the following script:

```
var wifi = require("wifi");
var ESP8266 = require("ESP8266");

function doFunc() {
  wifi.scan(function(err, aps) {
    console.log("--" + JSON.stringify(err) + "," + JSON.stringify(aps));
  });
}

setInterval(doFunc, 5000);
```
And run `process.memory()` and I see the number of free JsVars decrease over time.  This is obviously wrong and examination will begin now.

nkolban, 2015-10-31:
It was found that we had failed to execute `jsvUnLock()` against certain variables that had been allocated resulting in a substantial memory leak.  Examination was performed on which variables were not being unlocked and code changes made.  Tests were then conducted to ensure that the memory was released.  A soak test was then run for 30 minutes and the memory utilization remained constant over that period of time.

The fixes were committed and pushed.

closed 2015-10-31 by nkolban