espruino/Espruino #691: MQTT protocol

roccomuso, 2015-11-02:
Does it support MQTT communication?

gfwilliams, 2015-11-02:
Yes, although only as a client. There's a library, and a page on it here: http://www.espruino.com/MQTT

roccomuso, 2015-11-02:
ok, I'm gonna try it soon putting the espruino firmware on a NodeMCU devkit 1.0 (based on esp8266-12e). Have you ever tried this 'configuration'? should it works reliably?

gfwilliams, 2015-11-02:
You might be better asking on http://gitter.im/espruino/Espruino - I think someone had tried MQTT with it though.

roccomuso, 2015-11-03:
It seems that the MQTT module causes an error:
    ERROR: Out of Memory!
    Uncaught SyntaxError: Got EOF expected ','

gfwilliams, 2015-11-03:
I think you probably need something with a bit more memory than the ESP8266 then - like the Pico.

Perhaps you could create your own copy of the module and remove some functionality.

This isn't an Espruino bug though, so I'm closing this. If you have problems, maybe you could post [on the Forum](http://forum.espruino.com/)?

closed 2015-11-03 by gfwilliams