espruino/Espruino #698: Dedicated module for NeoPixel support

nkolban, 2015-11-09:
The NeoPixel (aka WS2811, WS2812 and others) is a smart LED that contains an embedded controller.  It can take a one wire stream of data and change its color based on this while at the same time allowing other NeoPixels to be serially connected in a chain.   In order to drive these NeoPixels, sub microsecond timings are required in the GPIO signals in conjunction with no-latency control (there is no clock signal).  This is extremely hard to do even in native C and believed to be unlikely to be achieved in a native JS app.  Possibilities for implementing these functions include hardware SPI and other techniques.

For some boards however, (eg. ESP8266) it may make sense to provide a native C/Assembler implementation that can then be wrapped in a JS module built into the base Espruino.  Proto types of this have been tested and seem to work.   The amount of compiled code appears to be on the order of 100-200 bytes with no apparent static storage being required and hence can live exclusively in flash memory without run-time impact on precious RAM.

This Issue tracks any work items needed to provide this function as well as discuss other ramifications such as design choices for the module and "correctness" of adding ostensibly non-JavaScript function into a JavaScript runtime environment.

nkolban, 2015-11-09:
Prior to work being performed on this issue, it is vital to read and study this whole thread ...

http://forum.espruino.com/conversations/276945

There appears to be some gold nuggets kept within that should be mined, considered, designed in and utilized.

nkolban, 2015-11-10:
During a commit today, some early work on this item has "leaked" into the master build.  Thankfully it is all consistent so no harm has been done.  The file called `targets/esp8266/jswrap_esp8266.c` exposes a static method called `neopixelWrite` on the module called `ESP8266`.  The signature of this function is:

    neopixelWrite(Pin, Array)

Where Pin is the Pin to use for data transmission and Array is an array of bytes that should be supplied as triples for red, green and blue.  It is unlikely that this will be the final selection for the interface but it gives us a starting point.

gfwilliams, 2015-11-11:
Can you look at the message I added to your accidental commit? There's some info in there that would really help to clean that function uo

nkolban, 2015-11-12:
While trying to implement the recipe supplied by @gfwilliams I mistakenly left in logic that was not needed.  The logic that we wanted to achieve was the use of `JSV_GET_AS_CHAR_ARRAY` to retrieve the data.  This was indeed used... however the logic to then walk the array and retrieve the pixels one at a time was not removed.  This has now been corrected.

nkolban, 2015-11-17:
The current implementation contains a guard that checks that the input data is an array ... the assumption here was that the input data should be an array of 3 bytes/pixel ... however some of the more efficient array types such as Uint8ClampedArray won't pass the check to see if we are a simple array as that produces false when the type is a special array (maybe jsvIsArrayBuffer or jsvIsTypedArray).

nkolban, 2015-11-24:
A recent change in the Makefile system resulted in the discovery of poor logic.  The NeoPixel functions must live in RAM and can't be in cache.  The correct solution is to prefix such functions with ICACHE_RAM_ATTR.  This has now been done.

nkolban, 2015-11-25:
The guarding of the array of data was loosened to allow Array and ArrayBuffer types.

gfwilliams, 2015-11-25:
We had talked about this - and you don't need any checks at all - the get_char_data does it all for you.

I'd mentioned that you can feed in `{data:[1,2,3],count:5}` and things like that.

nkolban, 2015-11-27:
Logic changed to remove all guards on input data with the assumption that the incoming JS object can be converted to an array of bytes.

tve, 2015-12-05:
I tweaked the neopixel code slightly and moved it back to flash due to space constraints. It's working fine for me and I don't see weird artifacts on the first bit. If someone sees problems, I have some ideas for how to fix it. Specifically: tweak the starting pointer and starting mask such that there's a pre-run of one iteration outputting 0 (i.e. no-op) to pre-load the code from flash before the second iteration outputs the first actual pixel.

tve, 2015-12-05:
I'm going to close this for now. Please reopen if issues reoccur.

nkolban, 2016-09-18:
I'm seeing timing issues that I can replicate and monitor with a logic analyzer.  The issues appear to be simply that the timings specified are not being properly honored compared to the original code that lived in flash.  Please let me know what I can do to document/describe.   When I attach a string of neopixels and try and alternate their colors ... they don't visualize correctly.   This is the core symptom that I then used a logic analyzer to hunt down the timings that were actually being generated.

tve, 2016-09-18:
Which version? Do you have screen shots of the logic analyzer that show the problem?

nkolban, 2016-09-18:
I am using the 1v87 build binaries downloaded from the Espruino web site.  I have also tried compiling the espruino/Espruino source directly from GIT and both exhibit exactly the same issues.

The script I am testing with looks as follows:

```

E.setClock(80);
var i = 0;
var esp8266 = require("ESP8266");
console.log(esp8266.getState());
pinMode(D2, "output");
function colorLeds(red, green, blue) {
  var data = [];
  for (var i=0; i<2; i++) {
    data.push(green);
    data.push(red);
    data.push(blue);
  }
  esp8266.neopixelWrite(NodeMCU.D2, data);
}

setInterval(function() {
  console.log("Changing ...!");
  i++;
  if ((i % 2) === 0) {
      colorLeds(255,0,0);
  } else {
      colorLeds(0,255,0);
  }
}, 1000);
```
which sends alternating red/green signals to a couple of NeoPixels.  I have a logic analyzer attached.  The data for which can be viewed using the Saleae logic analyzer found here ...

https://www.saleae.com/downloads

Attached to this post is a ZIP that contains a comma separated values of the recording that can be opened up in an editor (data.) and an input to Saleae (8 MHz, ...)

I am also available to take traces in real time or stream video if needed.  My suggestion to look at timings is to install the Saleae tool and load the exported data.

[NeoPixels.zip](https://github.com/espruino/Espruino/files/478819/NeoPixels.zip)

tve, 2016-09-18:
Why are you setting the clock to 80Mhz? I tuned both but 160Mhz is more precise.
Also, your test program works beautifully for me, except that I'm not sure why you have D2 in one place and NodeMCU.D2 in another, aren't those different pins? I changed both to D5 for my set-up.

nkolban, 2016-09-18:
I change my test script to be (and wired DIN of the pixels to D5):
```
var i = 0;
var esp8266 = require("ESP8266");
console.log(esp8266.getState());
function colorLeds(red, green, blue) {
  var data = [];
  for (var i=0; i<20; i++) {
    data.push(green);
    data.push(red);
    data.push(blue);
  }
  esp8266.neopixelWrite(NodeMCU.D5, data);
}

setInterval(function() {
  console.log("Changing ...!");
  i++;
  if ((i % 3) === 0) {
      colorLeds(255,0,0);
  } else if ((i%3) === 1) {
      colorLeds(0,255,0);
  } else {
      colorLeds(0,0, 255);
  }
}, 1000);
```
Unfortunately I get exactly the same symptoms as reported previously and shown in the video.  In your test environment:

* What kind of NeoPixels are you using?
* How many pixels are in your string/strand?
* What does a schematic of your environment look like?  Specifically, where are you drawing power for your pixels?
* Are you seeing determinicity over a period of time?   As the YouTube video shows, it I switched it on, we could appear to have a working strand for 20's of seconds before non-determinicty starts to creep in.

tve, 2016-09-18:
I believe I have the WS2812. 8 pixels, powered off 5v (from which there's a buck converter down to 3v). They're simply hooked up to the esp8266-12's D5 pin. I let the "walk" script I posted in the forum run for several minutes and other than my eyes hurting didn't see anything special.

nkolban, 2016-09-18:
I'm wondering if you ran solid color for the 8 pixels switching between green, red, blue rather than a rainbow ... we would spot any inconsistencies in your end.  Given that my setup is just one data point and that yours works ... I think we'll need some others to report in too.  I can recreate at will and would be delighted to run any further tests or show you live via Webcam.   I've spent a few days on this puzzle already but will try some additional things as well.  I'll see if I can't dig out a distinct NodeMCU and see if there might be something odd with this instance.

If we work through the possibilities ... then we have either bad signal being sent from ESP8266, bad signal arriving at LEDs or bad power to the LEDs (can't seem to think of anything else ... have swapped out LED strands and have tested same LED strands on Arduino test baseline).

A thought strikes me ... do we know "how" a NodeMCU GPIO output works?   And by this I mean does it use "pull ups or pull downs to reach signals?" ... for example, when LOW ... is it really driven to ground?  When HIGH is it really pulled to Vcc?  Does it use pull-ups or pull-downs to reach any of those?  I'm wondering out loud if there is an issue with the "nature" or "quality" of the signal produced as output by the ESP8266?   Again, these are just guesses ... but I'm open to trying and testing anything.

tve, 2016-09-18:
Running exactly your code, only `NodeMCU.D5` replaced by `D5` works great for me. I changed the interval to 200ms and it continues to work fine.

nkolban, 2016-09-19:
Again, many thanks for the feedback.  Given that I am having problems and your test environment isn't, then that speaks to a distinction between what you have and what I have.  I am not very amenable to the possibility that there is an issue my end in either my software, hardware, techniques or components.  Do you happen to have any suggestions on tests I might run that would allow me to make progress in the diagnostics?  I'm running low on things to try.

Since I am running my NodeMCU from a USB input and my NeoPixels from a 5V source (with common ground), I wondered if the voltage from the output of the NodeMCU to the DIN in the NeoPixels might have been an issue so I inserted a level shifter IC between NodeMCU and the NeoPixels.   Unfortunately that didn't seem to change a single aspects of the puzzle.   The abberations continued exactly as before with no apparent change in symptoms.

nkolban, 2016-09-19:
Closing at this time based upon the findings in the last post and the detailed story written up in this forum post:

http://forum.espruino.com/conversations/276945/?offset=75#comment13219383

closed 2016-09-19 by nkolban

gfwilliams, 2016-09-19:
> the timings specified are not being properly honored compared to the original code that lived in flash

Just IMO - but comparing with an original firmware before posting would have saved a lot of time here :)

nkolban, 2016-09-19:
I don't disagree at all ... I probably spent a good 6-8 hours in total chasing this one down.     I compounded my errors on top of each other.  Thinking to myself that this used to work "10 months ago" when I tested with it then I said "Well ... something changed" ... so I learned how to rebuild from the source, saw that the neoPixel code had been cleaned up and just to the conclusion that someone had broke something  (bad conclusion) ... so I then said to myself "Let me test again from the git source that was there 10 months ago".  So I replaced the NeoPixel code with what was there in the past and ... the timings that I got now from code that worked in the past no longer worked.  So I asked myself ... what changed ... and the biggest difference was running from flash as opposed to running from RAM.  So I jumped to the conclusion that running from flash had broke something.    It was hours later that I realize that Espruino on ESP8266 has changed the default CPU speed from 80MHz to 160MHz and hence the timings that were hard coded to work in the past were now no longer valid.

I reset and re-read what @tve had coded and, as is always the case, his diligence and care had covered all alternatives including 80 vs 160 MHz.   Once I had abandoned the train of thought that the code was in error, it was only then that I started looking for environmental issues.