espruino/Espruino #708: Implement analogWrite in ESP8266 port

nkolban, 2015-11-13:
As we study Espruino further, we have reached the point in investigating Pulse Width Modulation (PWM).  This appears to drive the hardware specific function called `jshPinAnalogOutput` which is not implemented in the ESP8266 port of Espruino.  The purpose of this Issue is to track a design, implementation and considerations to make that area of the function work.

nkolban, 2015-11-13:
Here is a start on important reference materials:

* ESP8266 SDK - PWM functions
* [How can I use PWM?](http://www.esp8266.com/viewtopic.php?f=6&t=545)

gfwilliams, 2015-11-14:
If you look at the STM32 jshardware.c you'll see Espruino can use `jstimer.c` for a built-in 'soft pwm' on pins that don't support it.

Assuming the util timer stuff actually works, that could be a nice easy way to get PWM :)

nkolban, 2015-11-15:
A first pass at PWM support through hardware was added and committed.  Unfortunately, it does not yet work ... (but since hardware PWM didn't work previously, I don't believe we have regressed).  When hardware PWM is engaged, we end up with a watchdog timer firing.  I don't know yet what might be going wrong.  A couple of suspicions:

1) Today, all executable code is linked into Flash as opposed to the default which is RAM.  This may very well include PWM handling which may be hardware interrupt driven ... which, if the case might be a problem.  The solution would be to keep the PWM handling in RAM ... but that opens up two issues.  The first is scarcity of RAM and the second is that we would have to change our build process to determine which modules should be excluded from the move from RAM to Flash.

2) There is a forum post here that seems to have the same issue.

http://www.esp8266.com/viewtopic.php?f=13&t=6458&p=33465

AkosLukacs, 2016-01-03:
Any progress on this?
Well, I guess the SDK uses hw timer, because according to the SDK docs (2C-ESP8266__SDK__Programming Guide__EN_v1.5.pdf) you can't use hw timer and pwm at the same time:

>## 8.4. PWM Related
> Herein only introduces the PWM related APIs in pwm.h. Users can inquire Espressif Systems for PWM documentation which will contain more details.
> PWM APIs can not be called when APIs in hw_timer.c are in use, because they use the same
hardware timer.

So i guess, either investigate if we can somehow work without the hw_timer functionality, or the esp8266 is soft-pwm only for espruino.

gfwilliams, 2016-01-04:
It sounds like PWM support may in fact be software only anyway?

Espruino itself has a soft PWM implementation based on the Utility timer. Since @tve has added the Utility Timer support (probably using hwtimer?) it should be pretty easy to just use that instead - it's just a few lines from `stm32/jshardware.c` that need to go in there - the only minor gotcha is the need to turn PWM off if someone does a subsequent digitalWrite/Read/etc.

gfwilliams, 2016-02-18:
Ditched hardware PWM for now as it was broken (ifdef'd out). Now using software PWM: https://github.com/espruino/Espruino/commit/cb0232b70e054133c3e3babe19dc675529b5a372

closed 2016-02-18 by gfwilliams