espruino/Espruino #710: Uint32Array not correct on ESP8266

jumjum123, 2015-11-16:
This only appears on ESP8266, Espruino boards are fine
var u = new Uint32Array(4); u[0] = 32; console.log(u,u[0]);
--> 0 // should be 32
var u = new Int32Array(4);u[0] = 32;console.log(u,u[0]);
--> 32 // here it is correct

nkolban, 2015-11-25:
I tested @jumjum123 story and it is exactly what he says it is.  I have no idea what is going on here ... but validation that there is something odd is half the battle.

gfwilliams, 2015-11-25:
Maybe check:

```
u = new Uint32Array(4); u[0] = 32; 
u.buffer
new Uint8Array(u.buffer);
(new Uint8Array(u.buffer))[3] = 32;
var u = new Uint32Array([32,0,0,0]);
u.buffer;
```

and add a bunch of debug to jsvArrayBufferSet and see what happens

MaBecker, 2016-12-11:
hmm added some debugs to  jsvArrayBufferIteratorDataToInt and came up with this change in src/jsvariterator.c

```
static JsVarInt jsvArrayBufferIteratorDataToInt(JsvArrayBufferIterator *it, char *data) {
  unsigned int dataLen = JSV_ARRAYBUFFER_GET_SIZE(it->type);
  JsVarInt v = 0;
  if (dataLen==1) v = *(int8_t*)data;
  else if (dataLen==2) v = *(short*)data;
  
  //else if (dataLen==4) v = *(int*)data;
  else if (dataLen==4) return *(int*)data;  // <-- MB

  else assert(0);
  if ((!JSV_ARRAYBUFFER_IS_SIGNED(it->type)))
    v = v & (JsVarInt)((1UL << (8*dataLen))-1);
  // jsiConsolePrintf("[jsvArrayBufferIteratorDataToInt: dataLen=%d,v=*(int*)data=%d, v=%d]\n",dataLen, *(int*)data, v); //MB
  return v;
}
```

test code and output
```
>u32 = new Uint32Array(4);
=new Uint32Array(4)
>u32[0] = 32;
=32
>u32.buffer
=new ArrayBuffer([32, 0, 0, 0, 0,  ... 0, 0, 0, 0, 0])
```

@gfwilliams - will this change  work for the other devices too ?

gfwilliams, 2016-12-12:
Do you have any other devices to test with?

I think this would be ok, but to be honest this isn't going to fix what happens when you put `0xFFFFFFFF` into your UInt32Array, because it won't say 0, but will say `-1`. Looks like this might need a more serious change - perhaps passing back a JsVar directly, which can handle the larger number range

MaBecker, 2016-12-12:
Yes, Pico and Puck.js - but no build env 

is this different from what you expected ?
>u32[0] = 0xFFFFFFFF;
=4294967295
>u32.buffer
=new ArrayBuffer([255, 255, 255, 255, 0,  ... 0, 0, 0, 0, 0])

gfwilliams, 2016-12-12:
No, that's great - what about if you do `Int32Array`? Then it should be -1

MaBecker, 2016-12-12:
it is 

```
>i32=Int32Array(4);
=new Int32Array(4)
>i32[0]= 0xFFFFFFFF;
=-1
```

gfwilliams, 2016-12-12:
Ok, looks good then - can you do a PR. Maybe a comment linking to this bug?

It still scares me that this fails on ESP8266.

gfwilliams, 2016-12-12:
... hit send too early.

```
 v = v & (JsVarInt)((1UL << (8*dataLen))-1);
```

should work.

1 is 32 bits, so if you shift it left by 32 you get 0 (it overflows). `0-1` is 0xFFFFFFFF and ANDing a 32 bit value with 0xFFFFFFFF should do nothing.

Unless I'm missing something, this sounds like compiler bug?

MaBecker, 2016-12-12:
As this issue is so old, I guess there is no need to hurry. 

Will write a small module to examine function returns of different types and try to setup a tool chain to build Puck.js on my site.

gfwilliams, 2016-12-12:
It's ok - I can't see it will cause a problem - we can always revert it if it does :)

gfwilliams, 2016-12-12:
Just to add, I think I know what's happening.

I reckon ESP8266 has a shift-left instruction that takes 5 bits (0..31) for the amount to shift. If you say `SHL 32` then it drops the 6th bit, leaving just 0. that means (`1<<0 == 1`, so `1-1 == 0` and it ANDs the value with 0).

IMO quite a worrying ESP8266 toolchain bug, but I guess fixing it would add extra instructions :(

MaBecker, 2016-12-12:
Flashed  the Pico with that travis build created by that merge and tested again - works as before !

So this issue can be closed - thanks for your support !

```
 _____                 _
|   __|___ ___ ___ _ _|_|___ ___
|   __|_ -| . |  _| | | |   | . |
|_____|___|  _|_| |___|_|_|_|___|
          |_| http://espruino.com
 1v89.969 Copyright 2016 G.Williams

>u32 = new Uint32Array(4);
=new Uint32Array(4)
>u32[0] = 0xFFFFFFFF;
=4294967295

>i32=Int32Array(4);
=new Int32Array(4)
>i32[0] = 0xFFFFFFFF;
=-1
 
```

gfwilliams, 2016-12-12:
Great - thanks!

closed 2016-12-12 by gfwilliams