espruino/Espruino #711: Would like to add STM32F103TB ...

jcw, 2015-11-16:
I'm trying to add a definition for this [HY-TinySTM103T](http://www.hotmcu.com/stm32f103tb-arm-cortex-m3-development-board-p-222.html) board, which as an STM32F103xB in a 36-pin package. The package type I'm using in the board .py file is VFQFPN36.

First off - I get a working build when using a serial port (yeay!). Now I'm trying to get USB working. The board has a discovery pin enabling circuit on A0, so I adjusted that, as well as the LED (and no button).

With USB, the binary size is 123088 (I'm building on Mac OSX, the size check in the makefile fails with "gdu not found").

But there's something odd with the auto-generated `jspininfo` files:

```
#define JSH_PIN_COUNT 80

#define JSH_PORTA_COUNT 16
#define JSH_PORTB_COUNT 16
#define JSH_PORTC_COUNT 16
#define JSH_PORTD_COUNT 16
#define JSH_PORTE_COUNT 16
#define JSH_PORTF_COUNT 0
```

Not sure this is related, but for some reason I can't get the USB device to appear. The LED (which is on A1), *does* light up as expected on reset (it's active low - I was able to toggle in in the USART build).

I can't quite figure out what's going on - some help would be appreciated :)

-jcw

gfwilliams, 2015-11-16:
It should be doable - it's STM32 ARM - so will be making the most of all available memory. The 128kB boards have some stuff cut out, but the Olimexino has SD card support, and this won't need that.

I'll have more of a think, but stuff to check out for now:

* The package name matches a column in the CSV file referenced from the BOARD.py - that might be why you have more pins than expected
* Check your GCC version. I've seen some issues with 4.9 and USB. Try 4.8 for now and see if it makes any difference...

jcw, 2015-11-16:
Yes! - this one works (from the Arduino IDE 1.6.5's Due package):
gcc version 4.8.3 20140228 (release) [ARM/embedded-4_8-branch revision 208322]

... while this one fails (installed from https://launchpad.net/gcc-arm-embedded):
gcc version 4.9.3 20150529 (release) [ARM/embedded-4_9-branch revision 227977]

Thank you Gordon, that was a great tip. I'll investigate the package / pin mixup.

gfwilliams, 2015-11-17:
Great! Glad you got it sorted :)

Maybe we'll close this when the package/pin thing gets sorted?

PR #713 (2015-11-17): Fix VFQFPN-36 package
closes #711
closed 2015-11-17 by gfwilliams