espruino/Espruino #722: ESP8266 getIP() only returns station IP

nkolban, 2015-11-22:
The function called getIP() only returns the station IP address.  This is a problem because the ESP8266 can also be an access point.  The suggestions here is to provide an optional argument to the function which can either be "station" or "accesspoint" to determine which IP address to return.  The default if not specified should be "station".

martinayotte, 2015-11-22:
In ArduinoESP, to distinguish both IPs, the functions are named localIP() and softAPIP().

gfwilliams, 2015-11-23:
Odd - I thought there had been some discussion about this before, but I can't find it now - I think rakeshpai mentioned moving to a `networkInterfaces` object, but that does seem a bit overcomplicated.

I guess another option is to have one as the default, but to return both, so:

```
{ 
  ip : ...,
  mac : ...,
  ap : { ip : ... },
  station : { ip : ... }
}
```

But to be honest the other solutions are good as well. I guess there could be a `getAPIP` or something that only ESP8266 devices implemented.

PS. In the Wiki where we discussed getIP's implementation, it used a callback so that it could be compatible with other network interface types. For instance with AT commands (GSM, ESP8266 module) you don't want to block while waiting for serial data to be returned. Same would apply for CC3000 too, if TI's drivers weren't so bad! :)

profra, 2015-11-23:
If station and AP are working both there are 2 IPs and 2 MACs... hence it would be more clear to ask them with two different functions... see reaction on AT command AT+CIFSR... picture...
https://espruino.microco.sm/api/v1/files/b4b12a58ea4823ec6fadd16911b7121fcf4770bf.png

nkolban, 2015-11-24:
Added a mac address to the returned data from getting the IP address.  Added a second function called getAPIP() that gets the Access Point information.

nkolban, 2015-11-24:
The documentation update for the new `getAPIP()` function as well as the documentation that the mac address is now included in the results was omitted.  This needs to be done before this issue should be closed.

nkolban, 2015-11-25:
There appears to be a function called "networkPutAddressAsString" that formats a Mac address ... this looks like exactly what we need to reuse in this area ... see the following forum post for more details....

http://forum.espruino.com/conversations/278075/#comment12646517

This needs added.

nkolban, 2015-11-26:
The docs were updated to include getAPIP.  In addition, the construction of the mac address string has now been changed to use the Espruino supplied string constructor.

closed 2015-12-05 by tve

trusktr, 2017-06-02:
I think the Espruino WiFi docs are outdated? But getIP does give IP when not in AP mode.

gfwilliams, 2017-06-02:
I'm not sure I understand - but you're commenting on the wrong bug again. These bugs are for Espruino running *on ESP8266 itself*, and I think you're talking about Espruino WiFi?

Please could you ask on the Espruino forums? http://forum.espruino.com/microcosms/1083/

And maybe give a bit more information? Which docs? What does it return?

The Espruino WiFi docs say:

```
/* Get the Access point's IP and MAC address and call 
callback(err, { ip : ..., mac : ...}). If err isn't null,
it contains a string describing the error
*/
exports.getIP = function (callback) { ... }
```

trusktr, 2017-06-08:
Hey, sorry for posting in the wrong place.

> `/* Get the Access point's IP and MAC address and call`

Not only for access point mode (docs only mention getting the IP in access-point mode), but it seems to work also for when connecting to another WiFi access point (which doc doesn't mention).

gfwilliams, 2017-06-08:
Just fixed with https://github.com/espruino/EspruinoDocs/commit/de4816197964d03f9e0f30cce9bb5e3d630c1563