espruino/Espruino #725: Make flash baud rate for ESP8266 configurable

nkolban, 2015-11-24:
The Makefile / build system for flashing the ESP8266 is currently hard-coded with a baud rate of 460800.  Unfortunately that is too high for some systems.  This request asks for this to be exposed externally as a parameter to the Makefile ... for example, a property called "BAUD" with a default of 115200.

gfwilliams, 2015-11-24:
The Makefile is pretty huge already - adding specific things for stuff like that does seem like overkill.

Why not set the default to 115200, and if someone wants more they can just modify the makefile themselves.

Honestly, searching for 115200 in the file and replacing it with 460800 is trivial compared with wading through documentation or the file trying to find out which environment variable you need to set to get the correct baud rate.

nkolban, 2015-11-24:
The default baud rate is now 115200 but can be optionally overriden with a FLASH_BAUD environment variable.

gfwilliams, 2015-11-24:
Sometimes I wonder why I bother writing anything when you post these bugs.

nkolban, 2015-11-24:
Look ... don't be vague ... say what you mean ... What have you written?  Where?  What is wrong with posting bugs? ... don't be vague ... we are all professional programmers here ... what is it you want to say?

gfwilliams, 2015-11-25:
Right after you posted the above, I posted the second comment - I thought it was pretty clear - I thought adding more options to an already complicated makefile was a bad move, and that anyone smart enough to figure out 115200 was too slow could open up the makefile, search for 115200 and replace it.

But 1 hour later, no discussion, but a commit that just did what you posted originally as if you hadn't seen my post at all. Hence my comment. It doesn't matter now, it's just a few lines - but it seems to happen quite a lot...

For instance just today you added the neopixel isArrayBuffer changes, but we spent half an hour on Gitter a few days ago talking about how get_char_array did all the checks so you didn't need any (and previous to that I also made comments to that effect on the original neopixel PR, [where I even wrote down the exact code you could use in my comment](https://github.com/espruino/Espruino/pull/692#issuecomment-155493451)).

I guess you probably have so many emails bouncing around you miss some - do you get emails when I post a reply to one of your bugs/PRs? It seems sometimes you might not. Or maybe you just scan-read and don't really take in what I'm saying - after all I know you're doing this in your free time, it's not like you have to do it, or any of the good work you've done on ESP8266 to date.

I don't want to dictate, "No, don't do this" because quite often you'll have good reasons, and generally I try and stay out of ESP8266 related stuff - but if I write something in response to one of the issues or PRs you posted (especially if it's got a section of code in it that I spent time writing), hopefully you can see it's a bit frustrating if it's totally ignored.

nkolban, 2015-11-25:
This is an excellent post.  Team's are complicated ... and virtual teams doubly so.  In my thinking, there can only ever be one leader ... and that leader is you.  You give an order or command, it must be followed.   Expect it to be questioned and argued ... but above all ... followed.  And that means that your instructions have to be explicit ... in this example ... you gave reasons as to why you thought it was a bad idea ... but you did not say "don't do it" or "hold off".  I expect that when you read that last sentence you are going to get mad ... but wait ... think it through for a moment.

The Makefile as it was in Github wasn't working for many of us (me especially).   The baud rate was 460800.  That baud rate works for me 5% of the time.   The baud rate of 230400 works about 60% of the time and 115200 works 100% of the time.  With the Makefile as it was, the liklihood of a new user downloading the project and trying to flash with that Makefile was not great.  As such, something had to change.  We could have replaced 460800 with 115200 and checked it back in ... end of story.  However, one of the most important things in collaborative teaming is not to "break" other folk's work.  I figured if the speed was set to 460800 by Mr @tve then he did that for a good reason.  For me to blindly change it to 115200 with no way for him to over-ride it would be arrogant and wrong.  So we have a compromise.  A speed of 115200 without ANYONE having to do anything special and an option to environmentally change it.

Now ...why was this change made without discussing it further with you?  A number of reasons:

1) It was urgent.  My thinking is always to end users.  An end user downloading the code base would be broken when trying to flash (this is arguable ... but lets assume that for just now).  In my philosophy of do no harm ... this was high.

2) I read that this was your suggestion ... but not your instruction.  I over-rode your suggestion.  I balanced this with a belief that a baud rate speed being exposed from a Makefile is standard practice in ESP8266 land.   I also balanced this with a work item that is in our queue to rework the whole Makefile subsystem when we can get to it.

The bottom line is to realize that you have resources out there ... folks like me ... who are at your disposal.  We ask for nothing except the pleasure of working on Espruino and moving the ball forward.  Since we can't go down to the pub at lunch time and have chicken in a basket nor can we have a good Chicken Madras on a Friday night ... friction will seem to appear ...  The trick is to realize that there are no hidden agendas.  I don't want anything from you :-) ... Not money, not my name in code, not public recognition .... to me ... this project is simply "fun" ... you have a simple choice ... is the hassle of interacting with others on the Espruino code base worth it?   If you think it is ... then understand frustrations will happen ... and when they do ... speak them loud and proud ... tell the person that you are frustrated at exactly what is on your mind ... and the air will clear.

Now ... for the future ... I fully expect that you eyeball pull requests ... if you see ones from me you don't like, reject them and tell me "Go fix this ..." ... simple as that ...

gfwilliams, 2015-11-26:
Thanks! Seems in this case you were a bit stuck in the middle.

In future, if it's not urgent and you don't agree, just tell me rather than doing it - it'd save a lot of trouble :)

But yes, if it's urgent then do whatever - just maybe put a note in the pull request saying that, so I know you actually read my comment :)

As you say, doing everything remotely makes things difficult sometimes - but it definitely helps to communicate as much as possible.

tve, 2015-11-26:
I don't want to add oil on the fire here, but I do appreciate the config variable. I always flash at 460800 baud and having to change the Makefile is easy on the one hand but sucks when it gets to make pull requests. I already have to carefully avoid committing the baud rate change in the boards file to a PR. These are just my 2c... :-)

closed 2015-12-05 by tve