espruino/Espruino #727: Fix use of jspNewObject in ESP8266 code ...

nkolban, 2015-11-24:
Within the ESP8266 port, there was usage of an Espruino function called "jspNewObject".  This function creates a new JS object and associates it with a logical named object type.  The use of this `jspNewObject` method was a bad choice and should be changed to use `jsvNewWithFlags(JSV_OBJECT)`.   It is likely that this was done in a number of places.  This Issue asks for the examination of the source to look for bad usage of `jspNewObject` and replace with the better function call.

nkolban, 2015-11-25:
Examination was made of the ESP8266 injected source files and references to jspNewObject were changed to jsvNewWithFlags.

gfwilliams, 2015-11-25:
Thanks - just to explain a bit: It's a lot faster to just create 'an object' - jspNewObject is really for if you have native functions that you've defined on some object and you want to return something that the user can call on that - using it when just returning data in response to a query isn't that useful, and it means that variables get allocated for the named Object as well as the one you wanted (it's probably at least 3) and memory is a bit tight on the ESP8266 :)

closed 2015-11-26 by nkolban