espruino/Espruino #728: esp8266 crashing

HyGy, 2015-11-24:
espruino console

```javascript
>
 ets Jan  8 2013,rst cause:4, boot mode:(3,0)
wdt reset
load 0x40100000, len 32412, room 16
tail 12
chksum 0xc1
ho 0 tail 12 room 4
load 0x3ffe8000, len 2140, room 12
tail 0
chksum 0xd8
load 0x3ffe8860, len 30096, room 8
tail 8
chksum 0x22
csum 0x22
s$Ãsl³Loading 6120 bytes from flash...
```

debugger console:

```javascript
>> intrHandlerCB
> jshGetWatchedPinState eventFlag=13
> ESP8266: jshPinGetValue pin=12, value=1
< jshGetWatchedPinState = 1
<< intrHandlerCB
Fatal exception 0(IllegalInstructionCause):
epc1=0x4023a07c, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

 ets Jan  8 2013,rst cause:4, boot mode:(3,0)

wdt reset
load 0x40100000, len 32412, room 16 
tail 12
chksum 0xc1
ho 0 tail 12 room 4
load 0x3ffe8000, len 2140, room 12 
tail 0
chksum 0xd8
load 0x3ffe8860, len 30096, room 8 
tail 8
chksum 0x22
csum 0x22

Restart info:
  reason:   2=exception
  exccause: 0
  epc1:     4023a07c
  epc2:     0
  epc3:     0
  excvaddr: 0
  depc:     0
Flash map 512KB:256/256, manuf 0xE0 chip 0x4016
Heap: 4976
Variables: 1023 @12ea = 12276bytes
Time sys=87046 rtc=1800991542
mode : sta(5c:cf:7f:01:29:64)
add if0
f r-1, > initDone
> jshInit
RTC read: 1448192567 1795156584 0x2fd90138 (0xdeadbeef)
RTC: restore sys=100757 rtc=1800993860
RTC: restored time: 1448192602 (delta=5837276 cal=5913us)
UStimer init
< jshInit
> jswrap_ESP8266_init
> jswrap_ESP8266WiFi_init
Wifi init, mode=1
< jswrap_ESP8266WiFi_init
< jswrap_ESP8266_init
> jshPinWatch: pin=5, shouldWatch=1
> ESP8266: jshPinSetState 5, GPIO_IN, pup=0, od=0
< jshPinWatch
> jshPinWatch: pin=12, shouldWatch=1
> ESP8266: jshPinSetState 12, GPIO_IN, pup=0, od=0
< jshPinWatch
> jswrap_ESP8266WiFi_init
Wifi init, mode=1
< jswrap_ESP8266WiFi_init
> jswrap_ESP8266WiFi_connect
>  - ssid=hygywlanpbj, password=nEgrO111
 - Current connect status: STATION_CONNECTING
scandone
f 0, < jswrap_ESP8266WiFi_connect
> jswrap_ESP8266WiFi_init
Wifi init, mode=1
< jswrap_ESP8266WiFi_init
< initDone
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
cnt 
chg_B1:-40

connected with hygywlanpbj, channel 6
dhcp client start...
Wifi connected to ssid hygywlanpbj, ch 6
>> intrHandlerCB
> jshGetWatchedPinState eventFlag=6
> ESP8266: jshPinGetValue pin=5, value=0
< jshGetWatchedPinState = 0
<< intrHandlerCB
>> intrHandlerCB
> jshGetWatchedPinState eventFlag=6
> ESP8266: jshPinGetValue pin=5, value=1
< jshGetWatchedPinState = 1
<< intrHandlerCB
>> intrHandlerCB
> jshGetWatchedPinState eventFlag=6
> ESP8266: jshPinGetValue pin=5, value=0
< jshGetWatchedPinState = 0
```

```javascript
process.env
={
"VERSION": "1v81.780",
"BUILD_DATE": "Nov 17 2015",
"BUILD_TIME": "22:46:57",
"GIT_COMMIT": "49cc77030a23f7756d3b57711a7e98c3ac67586a",
"BOARD": "ESP8266_BOARD",
"CHIP": "ESP8266",
"CHIP_FAMILY": "ESP8266",
"FLASH": 524288, "RAM": 81920,
"SERIAL": "35633a63-663a3766-3a30313a-32393a36-34",
"CONSOLE": "Serial1",
"EXPORTS": { "jsvLock": 1075912100, "jsvLockAgainSafe": 1075912328, "jsvUnLock": 1075913748, "jsvSkipName": 1075922636,
"jsvMathsOp": 1075928396, "jsvMathsOpSkipNames": 1075930060, "jsvNewFromFloat": 1075920376, "jsvNewFromInteger": 1075920280, "jsvNewFromString": 1075919928,
"jsvNewFromBool": 1075920344, "jsvGetFloat": 1075926136, "jsvGetInteger": 1075922368, "jsvGetBool": 1075926524, "jspeiFindInScopes": 1075941168,
"jspReplaceWith": 1075940792, "jspeFunctionCall": 1075949124, "jspGetNamedVariable": 1075943636, "jspGetNamedField": 1075944092, "jspGetVarNamedField": 1075944228,
"jsvNewWithFlags": 1075919684 }
}
```

example code (based on lawrencedon):

```javascript
require("ESP8266").logDebug(true);
var ESP8266WiFi=require('ESP8266');

var A0 = new Pin(5); // V-OUT 2  1um
var A1 = new Pin(12); // V-OUT 2.5um

var pm10 = 0;
var pm25 = 0;
var pm10_weight = 0;
var pm25_weight = 0;
var aqi = 0;
var P25Weight = 0;

setWatch(function(e) {
  var lowtime0 = (e.time - e.lastTime) * 1000;
  if (lowtime0 >= 10 && lowtime0 <= 90) {
//    console.log("pm10: " + lowtime0);
    pm10 += lowtime0;
  }
}, A0, {
  repeat: true,
  edge: "rising"
});

setWatch(function(e) {
  var lowtime1 = (e.time - e.lastTime) * 1000;
  if (lowtime1 >= 10 && lowtime1 <= 90) {
//    console.log("pm25: " + lowtime1);
    pm25 += lowtime1;
  }
}, A1, {
  repeat: true,
  edge: "rising"
});

setInterval(function(e) {
  pm10_weight = 0.30473 * Math.pow((pm10 / 30000) * 100, 3) - get * Math.pow((pm10 / 30000) * 100, 2) + 102.60291 * (pm10 / 30000) * 100 - 3.49616;

  pm25_weight = 0.30473 * Math.pow((pm25 / 30000) * 100, 3) - 2.63943 * Math.pow((pm25 / 30000) * 100, 2) + 102.60291 * (pm25 / 30000) * 100 - 3.49616;

  P25Weight = pm10_weight - pm25_weight;

  var pm10lowratio=pm10/3000;
  var pm25lowratio=pm25/3000;
  pm25lowratio=pm25lowratio-pm10lowratio;
  
  var concentrationpm25 = 1.1*Math.pow(pm25lowratio,3)-3.8*Math.pow(pm25lowratio,2)+520*pm25lowratio+0.62; // using spec sheet curve

  console.log('pm2.5: '+concentrationpm25/100 + ' pcs/0.01cf');
  console.log("PM1.0: "+pm10_weight + " --> " + pm10 + ' --> ' + pm10lowratio );
  console.log("PM2.5: "+pm25_weight + " --> " + pm25 + ' --> ' + pm25lowratio );

  pm10 = 0;
  pm25 = 0;

}, 30000);
```

nkolban, 2015-11-24:
Since we don't have a particularly good stacktrace mechanism ... when I hit problems like this I go for a "divide and conquer" approach. When I would do is place a "print" message at the entry and exit to each function. For example at the entry and exit of the setInterval function and the entry and exit of each of the setWatch functions. I would then let it run making sure that I am seeing the data I expect to appear in the console. When the failure happens, if all goes well ... we would see an entry into one function but not its corresponding exit. I would liberally place print statements throughout that failing function to "bracket" the statement that is failing.

Can you try that and lets see if we can pin it down some more.

HyGy, 2015-11-25:
```javascript
var ESP8266=require("ESP8266");
ESP8266.logDebug(true);

var A0 = new Pin(5); // V-OUT 2  1um
var A1 = new Pin(12); // V-OUT 2.5um

var pm10 = 0;
var pm25 = 0;
var pm10_weight = 0;
var pm25_weight = 0;
var aqi = 0;
var P25Weight = 0;

setWatch(function(e) {
  console.log('--> 1');
  var lowtime0 = (e.time - e.lastTime) * 1000;
  console.log('--> 2');
  if (lowtime0 >= 10 && lowtime0 <= 90) {
    console.log('--> 3');
//    console.log("pm10: " + lowtime0);
    pm10 += lowtime0;
    console.log('--> 4');
    
  }
}, A0, {
  repeat: true,
  edge: "rising"
});

setWatch(function(e) {
  console.log('--> 6');
  var lowtime1 = (e.time - e.lastTime) * 1000;
  console.log('--> 7');
  if (lowtime1 >= 10 && lowtime1 <= 90) {
    console.log('--> 8');
//    console.log("pm25: " + lowtime1);
    pm25 += lowtime1;
    console.log('--> 9');
    
  }
}, A1, {
  repeat: true,
  edge: "rising"
});

setInterval(function(e) {
  console.log('--> 10');
  pm10_weight = 0.30473 * Math.pow((pm10 / 30000) * 100, 3) - 2.63943 * Math.pow((pm10 / 30000) * 100, 2) + 102.60291 * (pm10 / 30000) * 100 - 3.49616;
  console.log('--> 11');

  pm25_weight = 0.30473 * Math.pow((pm25 / 30000) * 100, 3) - 2.63943 * Math.pow((pm25 / 30000) * 100, 2) + 102.60291 * (pm25 / 30000) * 100 - 3.49616;
  console.log('--> 12');

  P25Weight = pm10_weight - pm25_weight;
  console.log('--> 13');

  var pm10lowratio=pm10/3000;
  console.log('--> 14');
  var pm25lowratio=pm25/3000;
  console.log('--> 15');
  pm25lowratio=pm25lowratio-pm10lowratio;
  console.log('--> 16');
  
  var concentrationpm25 = 1.1*Math.pow(pm25lowratio,3)-3.8*Math.pow(pm25lowratio,2)+520*pm25lowratio+0.62; // using spec sheet curve
  console.log('--> 17');

  console.log('pm2.5: '+concentrationpm25/100 + ' pcs/0.01cf');
  console.log('--> 18');
  console.log("PM1.0: "+pm10_weight + " --> " + pm10 + ' --> ' + pm10lowratio );
  console.log('--> 19');
  console.log("PM2.5: "+pm25_weight + " --> " + pm25 + ' --> ' + pm25lowratio );
  console.log('--> 20');
  
  console.log('--> 23');
  pm10 = 0;
  console.log('--> 24');
  pm25 = 0;
  console.log('--> 25');

}, 30000);
```

```javascript
--> 1
--> 2
--> 3
--> 4
--> 1
--> 2
--> 6
--> 7
--> 6
--> 7
--> 8
--> 9
--> 6
--> 7
--> 1
--> 2
--> 6
--> 7
--> 6
--> 7
--> 8
--> 9
--> 1
--> 2
--> 1
--> 2
--> 1
--> 2
--> 6
--> 7
--> 1
--> 2
--> 1
--> 2
--> 1
--> 2
--> 1
--> 2
>
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)
load 0x40100000, len 1396, room 16
tail 4
chksum 0x89
load 0x3ffe8000, len 776, room 4
tail 4
chksum 0xe8
load 0x3ffe8308, len 540, room 4
tail 8
chksum 0xc0
csum 0xc0
2nd boot version : 1.4(b1)
  SPI Speed      : 80MHz
  SPI Mode       : QIO
  SPI Flash Size & Map: 32Mbit(512KB+512KB)
jump to run user1 @ 1000
don't use rtc mem data
sdÃslã¢þLoading 6095 bytes from flash...
```

tve, 2015-11-26:
I'm missing the context here. Can I use the code above to reproduce the issue or do I need special hardware attached to reproduce it?

tve, 2015-11-26:
Mhh, looking at the code I bet that it's an interrupt-code-in-flash problem. The GPIO state transition is captured by `intrHandlerCB` which needs the ICACHE_RAM_ATTR and should be very careful with what it calls. Specifically, it calls `jshPushIOWatchEvent`, which also needs this attribute and that calls ...
I don't know whether this call chain can all be placed into RAM. If not, a task needs to be posted, or something like that.

HyGy, 2015-11-26:
I think it is running successfull without any special hardware, it just detecting the 2 GPIO port low level percentage. And its read out every 30 seconds, then set it to 0.

tve, 2015-11-26:
From gitter chat:

>TvE: your code seems to calculate something from the pulse widths. What are "normal" pulse widths?

HyGy:
i'm using a dust sensor
pm2.5: 4.26179203685 pcs/0.01cf
PM1.0: 328.49002023166 --> 1025.26400000010 --> 0.34175466666
PM2.5: 1313.61349653432 --> 3491.70500000059 --> 0.82214700000
this is the output
@tve 2429.91119415283
this is the pm10 trough 30 sec
the first example this is 1025.264 and 3491.705
togther
63.36212158203
77.68082618713
38.38086128234
31.41021728515
65.24610519409
31.25715255737
62.93416023254
76.94292068481
45.66907882690
60.28985977172
43.21408271789
24.84107017517
35.41803359985
these are same raw values

for the pm10 variable when it measures once
as lowtime0

tve, 2015-12-03:
Would you mind testing again using the build at http://www.espruino.com/binaries/travis/7390e23a1e475a882edc9bbc20f0893947c7b925/espruino_1v83.tve_wifi_7390e23_esp8266.tgz
I'm hoping it will fix your problem, but beware that it breaks neopixelWrite().

HyGy, 2015-12-03:
I uploaded, its running, but I got this:

WARNING: Module "wifi" not found

This build using the old wifi api?

HyGy, 2015-12-03:
@tve It's seems to be stable now. I started it one hour ago. But that is interesting, that the measurements not the same as before. So I always got negative measurements (which is impossible).
I'll rewise my code, maybe I made something wrong with the calculations

closed 2015-12-03 by gfwilliams

tve, 2015-12-03:
module Wifi, capital W

tve, 2015-12-03:
I did something simple to measure positive and negative pulses of 160us and 800us respectively. The measurements I got were 4us high, so I declared victory. If you see issues, let me know. What I used is:
```
function meas(n) {
  var cnt = 0;
  var sum = 0;
  var w = setWatch(function(ev) {
      if (ev.lastTime === undefined) return;
      sum += ev.time-ev.lastTime;
      if (cnt === 0) console.log(ev);
      if (++cnt >= n) {
        clearWatch(w);
        console.log("Cnt=" + cnt, "width=" + (1000000.0*sum/cnt) + "us");
      }
    }, D5, {repeat: true, edge:'falling'}
  );
}
```

PR #750 (2015-12-03): esp8266 wifi library refactoring
This is a pretty large PR whose main piece is to refactor the wifi class. It now does everything requested in the wiki thread and more, has full documentation, and has functional tests in `./libs/network/esp8266/test`. (The tests need more docs so someone else can run them.)

This PR also attempts to fix setWatch by moving all the functions that run at interrupt time into static RAM. This has been briefly tested and is expected to fix #728. Unfortunately I had to remove neopixelWrite from static RAM due to space constraints and that broke it again according to  [... cut]