espruino/Espruino #737: instanceof doesn't think arrays are  objects

tve, 2015-11-27:
In Espruino:

```
>var a = [1,2]
=[ 1, 2 ]
>a instanceof Object
=false
```

in Chrome:

```
var a = [1,2]
undefined
a instanceof Object
true
```

gfwilliams, 2015-11-27:
Thanks! Just bodged this up for now.

Hopefully when this branch gets sorted everything will become a bit less hacky around there though: https://github.com/espruino/Espruino/tree/real_prototype_chain

closed 2015-11-27 by gfwilliams

opichals, 2022-04-09:
Hey @gfwilliams !

Still up to get this real_prototype_chain branch in? I have tried to rebase it on top of master, but lack the insight into the jsparse changes in which there are a lot of conflicts.

gfwilliams, 2022-04-19:
Thanks - yes, it's something I would still like to do if only to avoid the extremely slow symbol table lookup (which is just a bunch of 'if' statements). 

It's just been hard getting time for it, as nowadays most people seem to care more about how Bangle.js apps work than the actual internals at the moment.

I was sure there was an issue open for this but I don't see it, so I just filed a new one at https://github.com/espruino/Espruino/issues/2181

Do you think you could post up a link to what you've got on that issue? It's a while ago now but I might be able to remember what was going on.

commit 38249a51 (2015-11-27): Fix [] instanceof Object (fix #737)