espruino/Espruino #738: Function to convert a memory area to a String/ArrayBuffer

gfwilliams, 2015-11-27:
`E.memoryArea(addr, len) -> ArrayBuffer`

Where that ArrayBuffer directly accessed the underlying memory area. In fact `JSV_GET_AS_CHAR_ARRAY` should be able to use the pointer directly.

It's not of much use for RAM, but would be amazingly useful for Flash memory (sadly it wouldn't be ESP8266-friendly, but potentially it could be worked around).

Specifically this would be very useful for storing certificates/keys for HTTPS: http://forum.espruino.com/conversations/277780/?offset=25#12650582

gfwilliams, 2015-11-27:
For this to be properly useful we'd want jsvStringIterator to be able to handle it as well - this would allow some pretty cool things:

* code to be evaluated directly from flash
* waveforms to the played directly from flash

There must be a few other cases this would be useful in too.

tve, 2015-11-27:
+100 on code directly evaluated from flash!

Can you explain why it wouldn't be esp8266 friendly?

When you say "jsvStringIterator to be able to handle it as well", what does that mean? (I don't know much about that function.) Would that allow code to be executed from flash without loading it all into RAM? That would be awesome.

SpenceKonde, 2015-11-28:
It would be awesome if we could stuff a function into flash! This would do it, wouldn't it? - ie, function bigLongStaticFunction(a,b,c){return eval(E.memoryArea(0xF00D,4000));} // run 4000 byte long block of code from flash starting at address 0xF00D)?

gfwilliams, 2015-11-30:
Yes, something like that would work just fine. It could be useful!

gfwilliams, 2015-11-30:
`jsvStringIterator` just provides getChar/setChar/nextChar kind of functions - and hides the jumping around between jsvars that happens under the hood. Internally it has `currentVar`, `currentChar`, `varLength` fields, and they could be fiddled such that it would actually execute from flash.

> Can you explain why it wouldn't be esp8266 friendly?

`jslex.c` uses what's basically an inlined jsvStringIterator (modified slightly iirc to make it faster). Thing is it reads characters one at a time, which I didn't think would work with flash memory on ESP8266?

Potentially you could change both jsvStringIterator and jslex to read a word at a time, but it might hurt performance

gfwilliams, 2015-12-01:
`JSV_GET_AS_CHAR_ARRAY` needs to know about this as well, and a general grep for jsvGetFlatStringPointer might be an idea to see if there are other places we can use them.

commit 6ea0a47f (2016-02-08): Allow JSV_GET_AS_CHAR_ARRAY to get a pointer to memory in ArrayBuffers or memoryAreas. Fix #738
closed 2016-02-08 by gfwilliams