espruino/Espruino #740: `E.setBootUpCode`

gfwilliams, 2015-11-27:
Discussion here:

http://forum.espruino.com/conversations/275207/

Strangely it didn't seem to make it into a GitHub issue. Would be very easy with the addition of: https://github.com/espruino/Espruino/issues/738

gfwilliams, 2016-03-30:
So currently [the code here](https://github.com/espruino/Espruino/blob/master/src/jswrap_flash.c#L366) does:

* If memory[FLASH_MAGIC_LOCATION]!=FLASH_MAGIC, no code saved - return
* Check first uint32 of saved code. It tells us how much program data there is
* Decompress the code from flash into RAM

Ideally, there would be a second uint32 in the saved code that contains the length of a string in flash with code (as text) that would be executed at startup.

If that uint32 existed, at bootup (right before `onInit` was called) the textual code would be loaded into a memoryarea and executed.

Actually I'll take a quick look at this now...

nailxx, 2016-03-30:
Got it. I will think how we can do this.

Is it an existing issue on GitHub (#740?) or should we create new to discuss problems related to the feature?

gfwilliams, 2016-03-30:
It's an existing issue... will explain there....

gfwilliams, 2016-03-30:
Ok, try the `boot_code` branch.

You can do:

```
E.setBootCode("HELLO=42;all_my_js_code_here")
```

After doing that, Espruino won't actually reset. You'll have to do it with `reset()` or `load()`.

When you do reset, Espruino will execute all that JS code *as if it has just been uploaded*.

After that, to reset Espruino to a fresh state you'd need: `E.setBootCode("");reset()`

Not ideal, but it's a good start. I need to think a bit about the interaction with `save()` and `reset()`.

gfwilliams, 2016-03-31:
Branch deleted, merged into master.

closed 2016-03-31 by gfwilliams

nailxx, 2016-03-31:
Whoa! Works perfectly. Thank you for the feature and for merging it into master.

commit 3c76ff4f (2016-03-31): First hack at E.setBootCode (fix #740)
commit 92b8b58c (2016-03-31): `jsvEvaluate` now uses memory area for execution of JS strings (fix #817)
            Add `E.setBootCode` to allow JS scripts to be run without being in RAM (fix #740)