espruino/Espruino #743: Unable to flash Espruino to ESP8266

davidmoshal, 2015-11-30:
Hi, would appreciate help getting Javascript flashed onto the ESP8266.

Have been unable to install on [this board](http://www.aliexpress.com/item/LoLin-V3-NodeMcu-Lua-WIFI-Development-Board-NEW-VERSION/32524933087.html?spm=2114.01020208.3.2.VibewR&ws_ab_test=searchweb201556_1_79_78_77_80,searchweb201644_5,searchweb201560_10), using the following command:
```
cd  EspruinoBuilds/ESP8266
sudo python ../../esptool/esptool.py -bm qio -bz 512K -bf 40 -cp /dev/tty.wchusbserial1410 -cb 115200 -cd nodemcu -ca 0x00000 -cf "boot_v1.4(b1).bin" -ca 0x1000 -cf espruino_esp8266_user1.bin -ca 0x7E000 blank.bin
```

Results in this response:
```
usage: esptool [-h] [--port PORT] [--baud BAUD]
               {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,flash_id,read_flash,erase_flash}
               ...
esptool: error: argument --baud/-b: invalid arg_auto_int value: 'm'
```
Serial port is correct (NodeMCU was flashed there earlier).  
Location of esptool.py is correct too.
Board is marked `LoLin new NodeMCU v3`, and works just fine with Lua, reporting:
```
NodeMCU 0.9.6 build 20150704  powered by Lua 5.1.4
```
(which is the version I just flashed it with, prior to trying to flash the espruino build.

profra, 2015-11-30:
You need probably another flash tool... https://github.com/igrr/esptool-ck

nkolban, 2015-11-30:
@davidmoshal Mr @profra is 100% correct ... there are in fact TWO ESPTOOLS ....unfortunately, one is called esptool.py and one called ... well ... esptool.   In your first command, you are using the esptool from @profra's link.

davidmoshal, 2015-11-30:
@profra  and @nkolban: thanks, will do.

davidmoshal, 2015-11-30:
Wow, that worked!
Got an error from the Espruino IDE when sending to Espruino, but it still worked:
```
 1v83 Copyright 2015 G.Williams
WARNING: the esp8266 port is in beta, don't expect everything to work
Flash map 512KB:256/256, manuf 0xc8 chip 0x4016
WARNING: *** Your flash chip does not match your flash map ***
```

davidmoshal, 2015-11-30:
Firstly: I'm blown away by having a wifi-enabled javascript embedded board, for well under $10
-excellent work guys, very, very impressive!! 

The following script works just fine, and flashes across from the EspruinoIDE nicely.
(and it's great to have warnings in the IDE before flashing!)

```javascript
var wifi = require('wifi');
var http = require('http');

wifi.connect("<SSID>", "<PWD>", null, function(err, ip){
   print("err:",err,"ip:",ip);
   
  var srv = http.createServer(function(req, res){
    res.write("req" + JSON.stringify(req));
    res.end();
   });
   srv.listen(80);
  print("listening on port:", 80);
});
```  
However, now there are questions:

1. Where is the wifi module which is being used in `require('wifi')`?
   There doesn't appear to be one on the Espruino module page.
   Would be nice to know if anyone has tried websockets (or where to find info on how to try that).

2. For some reason the IP is printed in decimal.
   e.g:  "ip": 369754378
   interestingly, an online [decimal to ip converter](http://ncalculators.com/digital-computation/ip-address-hex-decimal-binary.htm), reports this as "22.10.1.10", when in fact the real IP is "10.1.10.22".

3. Apparently ports other than 80 are not supported?

4. Where can one find the most up-to-date documentation on using Javascript with the ESP8266
  -i.e: where are the actual modules documented, noting which functions work with which boards?

5. I have several different types of boards (Huzzah, Sparkfun, ESP-01, several ESP-12's from China, etc.) - I'd be willing to test the various modules on each board and report back if the documentation could be centralized somewhere (I realize you guys are working on actually getting the thing to work, and also have merged various projects into this one, but it would be useful to have a *canonical site of all things related to Espruino-ESP8266* !

Thanks again for the awesome work!

Dave
San Francisco, CA

nkolban, 2015-12-01:
Howdy @davidmoshal and first thanks for the kind words.   This is a team effort and there is a growing cast ... end user testing such as yours is a great thing and we consider you now part of the team.

The communal code base for the Espruino (including the port to ESP8266) can be found in Github.  There are a couple of branches of the code in flight just now but the latest docs that I know about can be found here:

https://github.com/espruino/Espruino/blob/master/targets/esp8266/docs/05-User_Guide.md

Though I suspect those may also be a little out of sync.

What I'd suggest doing is posting live to the gitter stream here:

https://gitter.im/espruino/Espruino

You may also find some docs currently being staged here ...

http://s3.voneicken.com/espruino/functions.html#t_Wifi

Don't forget to tune into the forums here:

http://forum.espruino.com/

davidmoshal, 2015-12-01:
Thanks @nkolban, I accept, I'm a fan - have been watching your videos and reading your book on and off for a couple of months now, great stuff. 

I'll take a look at the pointers you sent, with a view to helping the information converge to, hopefully, answer the end user's primary questions, namely: 

1. How do I interface this sensor/actuator/widget with each of these interactive IoT platforms:
  - Javascript on Pico + ESP8266,
  - Javascript on ESP8266 board X, 
  - Lua on ESP8266 board X (NodeMCU)

2. Similarly, how do I actually interface each device with each platform?
3. Is there a uniform device API across all platforms?

Take for example the situation with NeoPixels:
 - awesome for Education, wearables, etc.
Apart from Javascript on Gordon's boards, which handle them fine, it's not clear what subset of the other boards can do this. Even with the Espruino though, the API is a bit low level, potentially requiring hand-coded array manipulations.

i.e: raising the current level of abstraction would be great for the end-user!

David
ps: @nkolban, the December addition of your ESP8266 book is showing 'October' on the front cover :)

gfwilliams, 2015-12-01:
Just to add to this, @tve has spent ages working on this page: http://www.espruino.com/EspruinoESP8266

That's the one that will is user-visible and linked from the Reference page, so that should be the first stop for any information - if it's not then any updates to that are appreciated (details are near the footer of the page).

When the ESP8266 port is finally released as part of the distribution zip (a release or two from now hopefully) I'll update the reference with the wifi class. However at the moment the WiFi API is in flux so I'm keeping it off the main site to try not to confuse people.

davidmoshal, 2015-12-01:
That's the page I needed!
http://www.espruino.com/EspruinoESP8266
-thanks

tve, 2015-12-03:
>  1v83 Copyright 2015 G.Williams
>WARNING: the esp8266 port is in beta, don't expect everything to work
>Flash map 512KB:256/256, manuf 0xc8 chip 0x4016
>WARNING: *** Your flash chip does not match your flash map ***

You used the instructions for a 512KB flash module but your module has 4MB. No harm, but in the future this will prevent you from using features that take advantage of the extra flash...

I believe there's no "issue" left here and am closing the ticket. Please reopen if needed.

closed 2015-12-03 by tve

ceremcem, 2016-05-13:
FYI: I had the same problem because I had set the flash size to 4MB by passing `--flash_size 4m` (which was wrong, since it was in bits, so the correct option would be `--flash_size 32m`)

See esptool.py documentation's relevant section: https://github.com/themadinventor/esptool#flash-size---flash_size--fs