espruino/Espruino #753: esp8266 Wifi library doesn't handle NULL, null, undefined parameters correctly

tve, 2015-12-06:
Need to understand how missing, null, and undefined parameters show up and code accordingly. Specifically, startAP() doesn't handle `{password: null}` correctly and if the options are `null` or missing there's a problem too.

MaBecker, 2016-12-12:
Hi @tve, 

not sure if I understand this correctly.

This is how I understand startAP() 

* works without password
Wifi.startAP("TEST",{authMode:"open"},function(e){
  console.log("Started AP");
});

* for all other authMode a password is needed
Wifi.startAP("TEST",{authMode:"wpa_wpa2",password:"test1234"},function(e){
  console.log("Started AP");
});

Is that the point you where thinking about, when you created this issue ?

MaBecker, 2017-07-21:
startAP does not handle a call like this correct:
````
Wifi.startAP("TEST",{authMode:"open", password:null}, function(e){
    console.log("Started AP");
});
````
add a extra if to catch this.

closed 2017-07-21 by MaBecker