espruino/Espruino #754: esp8266: the WifiEventDetails object that gets created to deliver events does not get deallocated

tve, 2015-12-06:
In the wifi event handler a WifiEventObject gets created:
https://github.com/espruino/Espruino/blob/master/libs/network/esp8266/jswrap_esp8266_network.c#L1729
that then gets passed to the potential JS "on" handler. It gets unlocked at the end of the event handler, but that doesn't free it. The result is a global object that remains in the environment unnecessarily. How can this be freed?

gfwilliams, 2015-12-08:
You can't remove it.

I posted about this before, and Neil did a PR with a bunch of these calls removed a few weeks ago - the issue is:

```
JsVar *jsDetails = jspNewObject(NULL, "WifiEventDetails");
```

What it's doing IIRC is effectively:

```
function WifiEventDetails() {
}
jsDetails = new WifiEventDetails();
```

So jsDetails gets removed, but `WifiEventDetails` hangs around.

If you just want an object in which to return information, then create a simple Object with:

```
JsVar *jsDetails = jsvNewWithFlags(JSV_OBJECT);
```

That's pretty much like doing `jsDetails = {}`.

It's far faster, more efficient, and it's what's done everywhere else in Espruino when something needs to be returned - it only appeared in the ESP8266 stuff because I think Neil might have misunderstood what it was actually doing.

Generally I only use `jspNewObject` when I'm returning something that is supposed to have native functions on it (eg the HTTP connections). General data like that returned from `process.env` would just use `jsvNewWithFlags(JSV_OBJECT)`

tve, 2015-12-08:
Thanks for the explanation. I now remember seeing the discussion and I'll fix it :-).

commit 37c386d9 (2015-12-15): fix #754 (WiFiEventDetail object left around)
closed 2015-12-15 by tve