espruino/Espruino #756: fix esp8266 network re-init after reset

tve, 2015-12-07:
See Gordon's comment in http://forum.espruino.com/conversations/278564/#comment12666177

MaBecker, 2017-09-14:
@gfwilliams  is this the fix https://github.com/espruino/Espruino/commit/ee1fe51f1ba5c93c298c2cf23c344e5291dacc66 ?

gfwilliams, 2017-09-14:
It's not - but I think this issue was fixed a long time ago already?

The fix I put in there doesn't fix a known issue, but looking at the code I saw that if WiFi produced an event after being initialised, but before the JS interpreter had initialised, it could get into a really bad state. I think it might be worth testing on its own, and if it doesn't cause problems then we should keep it.

To be honest it's a shame the code doesn't just use JsVars to store the network state. It'd be much simpler and would mean we could use that whole 4k area that's currently used for just 256 bytes of network settings for saved code - bringing the size up to 16k.

gfwilliams, 2017-09-14:
Closing this bug - reading into the original forum post this definitely got fixed ages ago. Possibly in https://github.com/espruino/Espruino/commit/89d8c09b7e02c527203ccf8b07482a0fcde8a546

closed 2017-09-14 by gfwilliams