espruino/Espruino #757: Toggle esp8266's network_state

tve, 2015-12-07:
Toggle the network state on/off: network is online if in station mode and connected or if in ap mode.

MaBecker, 2016-12-11:
Hi @tve  is this still and issue ?

I use this commands to shutdown wifi 

require("Wifi").disconnect();
require("Wifi").stopAP();
require("Wifi").save();

>require("ESP8266").getState();
={
  "sdkVersion": "2.0.0(5a875ba)",
  "cpuFrequency": 160, "freeHeap": 9776, "maxCon": 10,
  "flashMap": "4MB:512/512",
  "flashKB": 4096,
  "flashChip": "0xe0 0x4016"
 }

tve, 2016-12-11:
Dunno, to be honest...

MaBecker, 2017-07-24:
@tve  maybe you where thinking of something like this:

wifi_set_opmode(NULL_MODE);

could be implemented as Wifi.disable() or Wifi.turnOff() ....

MaBecker, 2017-07-24:
So is there something missing in this output accept that saveMode is static (#752)?

    require("Wifi").getStatus();
    ={
       "mode": "sta",
        "station": "connected",
        "ap": "disabled",
        "phy": "11n",
        "powersave": "none",
        "savedMode": "off"
     }

Look complete to me.

MaBecker, 2018-01-06:
Hmm, so have something like this  @tve? 

````
var Wifi=require('Wifi');

/*
 do some wifi stuff
*/

// query the Wifi status variable that has the actual status 
if ( Wifi.status ) {
  // in station mode and connected or if in ap mode
  console.log('Wifi is up');
else {
  console.log('Wifi is down')
}

````

MaBecker, 2018-01-10:
@gfwilliams how to access  a global library variable from JS?

MaBecker, 2018-01-30:
like in I2C1

```
>  I2C1.setup({sda: D5, scl: D4} );
=undefined
>I2C1._options
={ "sda": D5, "scl": D4 }
```

gfwilliams, 2018-01-30:
Yes - you can add stuff to standard I2C as well. Just create a normal 'wrapped' class: https://github.com/espruino/Espruino/blob/master/src/jswrap_spi_i2c.c#L459

Any methods on that take a parent as a first argument, and then you just add to that: https://github.com/espruino/Espruino/blob/master/src/jswrap_spi_i2c.c#L557