espruino/Espruino #766: Print error message if input buffer gets full

gfwilliams, 2015-12-17:
We currently have `E.getErrorFlags` but a visible warning might save a bit of debugging time!

wilberforce, 2016-03-05:
Rather than having to poll  `E.getErrorFlags` could we set up an event listener, and register a function to call on error conditions?

Then things like low memory could be alerted at the time before space runs out!

gfwilliams, 2016-03-07:
I guess I could add an `E.on('error',function(flagName) {..})` sort of thing.

I wonder how much help it'd be? The events only get handled in the idle loop (they're not pre-emptive) so you're not going to be able to actually do anything about an out of memory event when it happens.

wilberforce, 2016-03-09:
At least you know it just after it happens, I've been adding process.memory() debug statements. I suppose the debugger in the ide does this.

commit 6a06417b (2017-08-16): Avoid printing error messages during execution, and report to console on idle when if errors were flagged (fix #766)
closed 2017-08-16 by gfwilliams