espruino/Espruino #770: More built-in code for ESP8266

gfwilliams, 2015-12-22:
Hi,

I just noticed that in the `Makefile` the ESP8266 only has USE_NET. I think at the least, `USE_GRAPHICS` might be handy for attaching graphics LCDs, and possibly `USE_FILESYSTEM` - which can allow you to connect an SD card to any available pins.

@tve I know you had concerns about the RAM usage, but I'd try it and see - this stuff is designed to statically allocate every little memory. I guess the only issue might the Flash usage?

MaBecker, 2016-01-09:
Hi,

added USE_GRAPHICS=1 and got a permanently reseting Espruino -  any idea ?

rS
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)
wdt reset
load 0x40100000, len 1396, room 16
tail 4
chksum 0x89
load 0x3ffe8000, len 776, room 4
tail 4
chksum 0xe8
load 0x3ffe8308, len 540, room 4
tail 8
chksum 0xc0
csum 0xc0
2nd boot version : 1.4(b1)
  SPI Speed      : 80MHz
  SPI Mode       : QIO
  SPI Flash Size & Map: 32Mbit(512KB+512KB)
jump to run user1 @ 1000

MaBecker, 2016-01-10:
using this exports 
export ESP8266_BOARD=1
export FLASH_4MB=1
export RELEASE=1
still rebooting but with a different cause
```
   __|___ ___ ___ _ _|_|___ ___
|   __|_ -| . |  _| | | |   | . |
|_____|___|  _|_| |___|_|_|_|___|
          |_| http://espruino.com
 1v84.MaBe_master_9aa07bc Copyright 2015 G.Williams
WARNING: the esp8266 port is in beta!
Flash map 4MB:512/512, manuf 0xe0 chip 0x4016
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)
load 0x40100000, len 1396, room 16
tail 4
chksum 0x89
load 0x3ffe8000, len 776, room 4
tail 4
chksum 0xe8
load 0x3ffe8308, len 540, room 4
tail 8
chksum 0xc0
csum 0xc0
2nd boot version : 1.4(b1)
  SPI Speed      : 80MHz
  SPI Mode       : QIO
  SPI Flash Size & Map: 32Mbit(512KB+512KB)
jump to run user1 @ 1000
r
```

MaBecker, 2016-01-31:
first success full try with  reduced vars (1000)

````
>process.memory();
={ "free": 971, "usage": 29, "total": 1000, "history": 0 }
>var g = Graphics.createArrayBuffer(16,8,24,{zigzag:true});
={
  "buffer": new ArrayBuffer(384)
 }
> 
````

MaBecker, 2016-02-01:
got 1310 vars but very litte heap

`````
>require("ESP8266").getState()
={
  "sdkVersion": "1.5.0",
  "cpuFrequency": 160, "freeHeap": 600, "maxCon": 10,
  "flashMap": "4MB:512/512",
  "flashKB": 4096,
  "flashChip": "0xe0 0x4016"
 }
````
any suggestion how to drop/shrink fonts ?

gfwilliams, 2016-02-01:
You could instead just modify their definition to fit into Flash memory instead?

MaBecker, 2016-02-01:
not sure how too, any hint ?

now running with a compromise of 1000 vars 

`````
 _____                 _
|   __|___ ___ ___ _ _|_|___ ___
|   __|_ -| . |  _| | | |   | . |
|_____|___|  _|_| |___|_|_|_|___|
          |_| http://espruino.com
 1v84.MaBe_master_f35ac96 Copyright 2016 G.Williams
WARNING: the esp8266 port is in beta!
Flash map 4MB:512/512, manuf 0xe0 chip 0x4016
>require("ESP8266").getState()
={
  "sdkVersion": "1.5.0",
  "cpuFrequency": 160, "freeHeap": 9560, "maxCon": 10,
  "flashMap": "4MB:512/512",
  "flashKB": 4096,
  "flashChip": "0xe0 0x4016"
 }
````

MaBecker, 2016-02-01:
that is  the reason why I am so keen on having GRAPHIC

[image]

gfwilliams, 2016-02-01:
Looks like you need something a bit like this: https://github.com/espruino/Espruino/blob/master/src/jsutils.h#L62

Ideally you'd make a new `#define` in `jsutils.h` that just applied `__attribute__((section(".irom.literal")))` for ESP8266 and did nothing for other platforms, then you'd make sure you use the `READ_FLASH_UINT8` macro whenever you were reading from the font.

MaBecker, 2016-02-01:
ok, extended jsutil.h see below

can you please name a file and line where and how to use `READ_FLASH_UINT8`  and how to upload the fonts.

````
#if defined(ESP8266)
// For the esp8266 we need to add CALLED_FROM_INTERRUPT to all functions that may execute at
// interrupt time so they get loaded into static RAM instead of flash. We define
// it as a no-op for everyone else. This is identical the ICACHE_RAM_ATTR used elsewhere.
#define CALLED_FROM_INTERRUPT __attribute__((section(".iram1.text")))

//-- MaBecker - Begin ---------------------
/** Place constant strings into flash when we can in order to save RAM space. Strings in flash
    must be accessed with word reads on aligned boundaries, so we'll have to copy them before
    regular use. */
#define FLASH_STR(name, x) static char name[] __attribute__((section(".irom.literal"))) __attribute__((aligned(4))) = x

/// Get the length of a string in flash
size_t flash_strlen(const char *str);

/// Copy a string from flash to RAM
char *flash_strncpy(char *dest, const char *source, size_t cap);

/** Read a uint8_t from this pointer, which could be in RAM or Flash.
    On ESP8266 you have to read flash in 32 bit chunks, so force a 32 bit read
    and extract just the 8 bits we want */
#define READ_FLASH_UINT8(ptr) ({ uint32_t __p = (uint32_t)(char*)(ptr); volatile uint32_t __d = *(uint32_t*)(__p & (uint32_t)~3); ((uint8_t*)&__d)[__p & 3]; })
//-- MaBecker - End -----------------------

#else
#define CALLED_FROM_INTERRUPT
#endif
````

gfwilliams, 2016-02-02:
Not quite sure I understand what you've done there - isn't that pretty much what's in jsutils already?

I meant, something like:

```
#ifdef ESP8266
#define IN_FLASH_MEMORY __attribute__((section(".irom.literal")))
#else
#define IN_FLASH_MEMORY
#endif
#define READ_FLASH_UINT16(ptr) ((READ_FLASH_UINT8(ptr)<<8) | READ_FLASH_UINT8(ptr+1) )
```

Look in `https://github.com/espruino/Espruino/tree/master/libs/graphics` for fonts. For example:

https://github.com/espruino/Espruino/blob/master/libs/graphics/bitmap_font_3x5.c

```
const unsigned short LCD_FONT_3X5[LCD_FONT_3X5_CHARS*5] IN_FLASH_MEMORY = {
// ...
void graphicsDrawChar4x6(JsGraphics *gfx, short x1, short y1, char ch) {
  int idx = ((unsigned char)ch) - 33;
  if (idx<0 || idx>=LCD_FONT_3X5_CHARS) return; // no char for this - just return
  int cidx = idx % 5;
  idx -= cidx;
  int y;
  for (y=0;y<6;y++) {
    unsigned short line = READ_FLASH_UINT16[&LCD_FONT_3X5[idx + y]] >> (cidx*3);
    if (line&1) graphicsSetPixel(gfx, x1+0, y+y1, gfx->data.fgColor);
    if (line&2) graphicsSetPixel(gfx, x1+1, y+y1, gfx->data.fgColor);
    if (line&4) graphicsSetPixel(gfx, x1+2, y+y1, gfx->data.fgColor);
  }
}
```

for rendering of the vector font you have to look into graphics.c though

MaBecker, 2016-02-02:
sorry, just starting with stuff like this, hope to improve day by day - thanks 

like to check result with  `find_big_ram.sh` and `find_big_rom.sh` 
how to create the input put file named espruino_XXXX.lst for this tools ?

`make topstrings` is broken on MAC OS X, will  fixit

MaBecker, 2016-02-02:
changed code in `jsutils.h`  and  `bitmap_font_4x6`  

DISABLE_LTO=1 make      

used `xtensa-lx106-elf-objdump  -s ./libs/graphics/bitmap_font_4x6.o`to create this text files 

[version_rodata.txt](https://github.com/espruino/Espruino/files/114184/version_rodata.txt)

[version_irom_literal.txt](https://github.com/espruino/Espruino/files/114186/version_irom_literal.txt)

@gfwilliams Why is the contents of section .rodata or .irom.literal using so much space filled with 0x00 ?

gfwilliams, 2016-02-02:
I have no idea - it could be zero-initialised structures? I'd have hoped that GCC would be smart enough to group them together in RAM and zero-init them with a loop, but I don't remember seeing it do that in ARM and I doubt it does for Xtensa.

MaBecker, 2016-02-03:
v1: `const unsigned short LCD_FONT_4X6[LCD_FONT_4X6_CHARS*6] IN_FLASH_MEMORY = { ...}`

v2: `const unsigned short LCD_FONT_4X6[] IN_FLASH_MEMORY = { ...}`

result: v2 shrinks irom.literal  ;-)

[bitmap_font_4x6.o.irom.literal_v1.txt](https://github.com/espruino/Espruino/files/115633/bitmap_font_4x6.o.irom.literal_v1.txt)

[bitmap_font_4x6.o.irom.literal_v2.txt](https://github.com/espruino/Espruino/files/115634/bitmap_font_4x6.o.irom.literal_v2.txt)

gfwilliams, 2016-02-03:
Wow, thanks! So I was basically allocating 5x more memory than needed! I guess all those fonts could be fixed in a similar way.

MaBecker, 2016-02-03:
this is what the latest Makefile from espruion/ESPRUINO uses for GRAPHICS:

```
ifdef USE_GRAPHICS
DEFINES += -DUSE_GRAPHICS
INCLUDE += -I$(ROOT)/libs/graphics
WRAPPERSOURCES += libs/graphics/jswrap_graphics.c
SOURCES += \
libs/graphics/bitmap_font_4x6.c \
libs/graphics/graphics.c \
libs/graphics/lcd_arraybuffer.c \
libs/graphics/lcd_js.c
```

changed _4x6 to _3x5.c and started ``make``

**holy moly** const unsigned short LCD_FONT_3X5 this full off bugs - fixed and yes  same result !

MaBecker, 2016-02-03:
- default bitmap_font_4x6 is now in stored in rom

will now start some test

MaBecker, 2016-02-03:
decided to test ram font first, but didn't look right ?!

bitmap_font_3x5.c  
```
const unsigned short LCD_FONT_3X5[] IN_FLASH_MEMORY = { // from 33 up to 127
    PACK_5_TO_16( _X_ , X_X , _X_ , _X_ , X_X ), // !"#$%
```
and bitmap_font_4x6.c

```
const unsigned short LCD_FONT_4X6[]  = { // from 33 up to 127
    PACK_5_TO_16( _X_ , X_X , X_X , _X_ , X_X ),
```
are setup with the same entries. 

Gordon, please double check.

MaBecker, 2016-02-04:
tried vector font, result looks strange

```
// connected to a 8x8 neopixel plate on pin D5
var esp = require('ESP8266');
var pin = D5;

var leds = Graphics.createArrayBuffer(8,8,24,{zigzag:true}); 
leds.flip = function() { esp.neopixelWrite(pin,leds.buffer); };

leds.clear();
leds.setFontVector(5);
leds.setColor(0.2,0,0);
leds.drawString("!",0,0);
leds.flip();
```
[image]

Any idea ?

MaBecker, 2016-02-04:
found it, something wrong with the initial setting of zigzag - no idea where to change that

`````
var leds = Graphics.createArrayBuffer(8,8,24,{zigzag:false});
leds.flip = function() { esp.neopixelWrite(pin,leds.buffer); };

leds.clear();
leds.setFontVector(5);
leds.setColor(0.2,0,0);
leds.drawString("!",0,0);
leds.flip(); 
`````
[image]

gfwilliams, 2016-02-04:
> something wrong with the initial setting of zigzag

I'm not sure what you mean? zigzag is for displays with the pixels wired in a zig-zag pattern - like the RGB123. The effect you're showing is what you'd expect if you wired it to a display that wasn't zig-zagged.

> bitmap_font_3x5.c and bitmap_font_4x6.c are setup with the same entries.

I don't think so? One is 6px deep, the other is 5px deep. Both are 3px wide (excluding the space between chars) which is a bit confusing though.

MaBecker, 2016-02-04:
so font 4x6 : 3x5 in 4x6  - got that

my neopixel plate is zigzag wired like this 

00 01 02 03 04 05 06 07
09 10 11 12 13 14 15 16
...... 

to address the pixel correctly I have to set zigzag option to **false,**  what  is in my understanding wrong compared with the docs.

gfwilliams, 2016-02-04:
> so font 4x6 : 3x5 in 4x6 - got that

If you look, you'll see that the 4x6 font does use the bottom row occasionally, so it isn't just a 3x5 font with the bottom row left empty.

```
00 01 02 03 04 05 06 07
09 10 11 12 13 14 15 16
```

Yes, that's an entirely normal scanline-style layout. How would you lay out a grid of pixels like that in a way that was less zig-zaggy??

`zigzag` is for:

```
00 01 02 03 04 05 06 07
15 14 13 12 11 10 09 08
...
```

gfwilliams, 2016-02-04:
Also, the docs say `zigzag = whether to alternate the direction of scanlines for rows` - I think that makes sense in this case?

MaBecker, 2016-02-04:
thanks for enlighten me !

MaBecker, 2016-02-08:
using font stored in irom will reboot if using drawSting()

downloaded yesterdays source version

found macro READ_FLASH_UINT8 in src/jsutils.h, added
````
#ifdef ESP8266
#define IN_FLASH_MEMORY __attribute__((section(".irom.literal")))
#else
#define IN_FLASH_MEMORY
#endif
````

tried to use it like this in file lib/graphics/bitmap_font_4x6.c
```
//const unsigned short LCD_FONT_4X6[]  = { // from 33 up to 127
const unsigned short LCD_FONT_4X6[] IN_FLASH_MEMORY = { // from 33 up to 127
......
    //int line = LCD_FONT_4X6[idx + y] >> (cidx*3);
    int line = READ_FLASH_UINT8(LCD_FONT_4X6[idx + y]) >> (cidx*3);
````
ESP still reboots, what am I missing ....
 
@gfwilliams  I need some help on this

gfwilliams, 2016-02-08:
Why didn't you use the code [I posted above](https://github.com/espruino/Espruino/issues/770#issuecomment-178441225)? LCD_FONT_4X6 is 16 bits, so you'd need to make and use  a READ_FLASH_UINT16.

I don't know whether that would help, but it would be a good start.

MaBecker, 2016-02-08:
changed ( wood tree thing ) - same error

````
var leds = Graphics.createArrayBuffer(8,8,24); 
leds.flip = function() { esp.neopixelWrite(pin,leds.buffer); };
leds.off = function(){leds.clear();leds.flip();};
var string = function() {
  leds.clear();
  leds.setColor(0.2,0,0);
  leds.drawString("X",0,0);
  leds.flip();  
};

>string();
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)
load 0x40100000, len 1396, room 16
tail 4
chksum 0x89
load 0x3ffe8000, len 776, room 4
tail 4
chksum 0xe8
load 0x3ffe8308, len 540, room 4
tail 8
chksum 0xc0
csum 0xc0
2nd boot version : 1.4(b1)
  SPI Speed      : 80MHz
  SPI Mode       : QIO
  SPI Flash Size & Map: 32Mbit(512KB+512KB)
jump to run user1 @ 1000
don't use rtc mem data
````

had to rename to .txt to allow upload
[jsutils.c.txt](https://github.com/espruino/Espruino/files/121976/jsutils.c.txt)
[bitmap_font_4x6.c.txt](https://github.com/espruino/Espruino/files/121977/bitmap_font_4x6.c.txt)

@gfwilliams  I need more help with this

MaBecker, 2016-02-10:
@gfwilliams is there a test/sample code to proof that READ_FLASH_UINT8(ptr) is working correct ?

gfwilliams, 2016-02-10:
@JumJum tried it. It's used all the time when executing code, but you would
have to try E.memoryArea on flash memory to see if it crashed or not

MaBecker, 2016-02-10:
Ok, will try to implement a simple lib with a array stored in irom and some prints 
and them hopefully figure out what is going wrong.

gfwilliams, 2016-02-10:
Thanks. Actually peek8 uses it. Just try and access any address in irom

MaBecker, 2016-02-10:
true, can use peek, is much easier

there are two macs stored in flash for each ESP12:
````
var ESP_OTP_MAC0   = 0x3ff00050;
var ESP_OTP_MAC1   = 0x3ff00054;
peek32(ESP_OPT_MAC0).toString(16);
peek32(ESP_OPT_MAC1).toString(16);
````
and compare it with
```` 
require("Wifi").getAPDetails();
require("Wifi").wifi.getIP();
`````

this what I will do first ;-)

MaBecker, 2016-02-10:
macros  READ_FLASH_UINT16(ptr) works fine if ptr address is good for a 32bit read (ptr % 4 = 0),

@gfwilliams I have no clue how to fix this

`````
// Company  Espressif Inc.
// OUI 5C-CF-7F
// MAC range 5C-CF-7F-00-00-00 - 5C-CF-7F-FF-FF-FF
// mac of my ESP 5c:cf:7f:05:c1:00
// fixed part at 0x3ff0005c
// variable part at 0x3ff00054
("00000000"+peek32(0x3ff0005c).toString(16)).slice(-8);
="005ccf7f"
("00000000"+peek32(0x3ff00054).toString(16)).slice(-8);
="020005c1"

// first two bytes - ok
("0000"+peek16(0x3ff00054).toString(16)).slice(-4);
="05c1"

// second two bytes - not ok
>("0000"+peek16(0x3ff00056).toString(16)).slice(-4);
="0000"

// first byte - ok
("00"+peek8(0x3ff00054).toString(16)).slice(-2);
="c1"

// second byte - not ok
("00"+peek8(0x3ff00055).toString(16)).slice(-2);
="00"

````

gfwilliams, 2016-02-11:
Ok, thanks - so looks like READ_FLASH_UINT8 doesn't actually work after all that :( 

You'd have to maybe look at the disassembled code and see what's happening. Maybe it's getting optimised away. There is the `volatile` keyword there, but you could try adding it to the pointer casting as well, and see if that helps.

jumjum123, 2016-02-12:
Just tried reading from 0x3ff00054 using peek32, peek16, peek8. Everything looks fine to me:
peek32(0x3ff00054).toString(16)
="200991a"
>peek16(0x3ff00054).toString(16)
="991a"
>peek16(0x3ff00056).toString(16)
="200"
>peek8(0x3ff00054).toString(16)
="1a"
>peek8(0x3ff00055).toString(16)
="99"
>peek8(0x3ff00056).toString(16)
="0"
>peek8(0x3ff00057).toString(16)
="2"

MaBecker, 2016-02-12:
Thanks, jumjum123 !
please let me know what this is base on, eg. like version and define macros used

gfwilliams, 2016-02-12:
Hmm. Any ideas what is different? I could be that somehow @MaBecker isn't compiling with the right flags, so the #define `READ_FLASH_UINT8` ends up being the non-ESP8266 one?

jumjum123, 2016-02-12:
base is 
- git clone https://github.com/espruino/Espruino.git
- curl -Ls http://s3.voneicken.com/xtensa-lx106-elf.tgx | tar Jxf -
- curl -Ls http://s3.voneicken.com/esp_iot_sdk_v1.5.0.tgx | tar Jxf -
compiling with 
ESP8266_BOARD=1 make

MaBecker, 2016-02-12:
on my way .. this is similar to what I do building on MAC OS accept this

added missing lib from espressif:

    liblwip_536_for_ESP8266_NONOS_SDK_V1.5.0.zip

exports:

export FLASH_4MB=1
export ALT_RELEASE=MaB

**cool, now it is working !**

now adding GRAPHIS and check again

still working, so I keep on moving font to flash on this version

MaBecker, 2016-02-15:
using leds.drawString("X",0,1); have diff results:
RAM : fine
FLASH: wrong

what about this second macro named in [E.memoryArea not working on ESP8266 #23](http://forum.espruino.com/comments/12794772/) ?

@gfwilliams:  Can it be  used inside macro "READ_FLASH_UINT8(ptr)" by checking ptr ?

gfwilliams, 2016-02-15:
@MaBecker yes, that was the idea. However you don't want to modify `READ_FLASH_UINT8(ptr)` since that would make it very slow to execute code. It would be better to create a new macro which contains the extra code as well.

MaBecker, 2016-02-18:
I was trying to ceate a new macro, but without success. 

Can someone please write this few lines of code for that macro  that follows  
````
//if (addr < RAM+ADDR) addr|=0x40200000;"
#define ESP_READ_FLASH_UNIT8(ptr) ........
`````
so I can use 
````
#define ESP_READ_FLASH_UINT16(ptr) ((ESP_READ_FLASH_UINT8(ptr)<<8) | ESP_READ_FLASH_UINT8(ptr+1) )  
`````
to read bitmap font from flash

gfwilliams, 2016-02-19:
I'd just stick it in READ_UINT8 for now:

```
#define READ_FLASH_UINT8(ptr) ({ uint32_t __p = (uint32_t)(char*)(ptr); if (__p < RAM+ADDR) __p|=0x40200000; volatile uint32_t __d = *(uint32_t*)(__p & (uint32_t)~3); ((uint8_t*)&__d)[__p & 3]; })
```

It's not perfect but would be good enough to see if it works.

MaBecker, 2016-02-19:
thanks, used like this  in my build 

what to use for "RAM+ADDR" ?

jsutils.c
````
#define ESP_READ_FLASH_UINT8(ptr) ({ uint32_t __p = (uint32_t)(char*)(ptr); if (__p < RAM+ADDR) __p|=0x40200000; volatile uint32_t __d = *(uint32_t*)(__p & (uint32_t)~3); ((uint8_t*)&__d)[__p & 3]; })
#define ESP_READ_FLASH_UINT16(ptr) ((ESP_READ_FLASH_UINT8(ptr)<<8) | ESP_READ_FLASH_UINT8(ptr+1) )    
````
and in bitmap_4x6.c
`````
#if defined(ESP8266)
    unsigned short line = ESP_READ_FLASH_UINT16(&LCD_FONT_4X6[idx + y]) >> (cidx*3);
#else
    unsigned short line = READ_FLASH_UINT16(&LCD_FONT_4X6[idx + y]) >> (cidx*3);
#endif  
`````

gfwilliams, 2016-02-19:
Try `0x400000` - @tve - any thoughts here? That's just the max address of flash available at the moment, rather than the RAM base address.

MaBecker, 2016-02-20:
no, not getting better.... still get unexpected result for drawSting.

Will add os_printf with %p and then check with peek, because i think reading from irom.literal on esp is different than reading from ram.

gfwilliams, 2016-02-20:
But it's not crashing? Maybe try:

```
#define ESP_READ_FLASH_UINT16(ptr) ((ESP_READ_FLASH_UINT8(ptr+1)<<8) | ESP_READ_FLASH_UINT8(ptr) )    
```

I just guessed on the byte order.

Also, you could try keeping the array in RAM, but trying to access it with the existing code - it'd help to narrow it down

MaBecker, 2016-02-20:
looks like I can nail it down:

using this print statement to get details when using "drawString("!",0,0);"
````
printf("%p:idx=%d,y=%d,cidx=%d,line=%X\n",  &LCD_FONT_4X6,idx,y,cidx,x1,y1,line);
````

expected values for line: 
[0]=556a
[1]=556a
[2]=1fea
[3]=2d42
[4]=47c0
[5]=5f42

and got:
0x40264716:idx=0,y=0,cidx=0,line=6AEA
0x40264716:idx=0,y=1,cidx=0,line=EA42
0x40264716:idx=0,y=2,cidx=0,line=42C0
0x40264716:idx=0,y=3,cidx=0,line=C042
0x40264716:idx=0,y=4,cidx=0,line=4200
0x40264716:idx=0,y=5,cidx=0,line=96

MaBecker, 2016-02-20:
next test, build first six chars ins console and compare to peek16()

`````
// [offset] = <value of PACK_5_16>, <peek16(addr+offset);
>first_six_chars();
[0]=556a,556a
[1]=556a,ea55
[2]=1fea,1fea
[3]=2d42,421f
[4]=47c0,2d42
[5]=5f42,c02d
[6]=400,47c0
[7]=2496,4247
[8]=7316,5f42
[9]=5303,5f
[a]=306,400
[b]=487,9604
`````

MaBecker, 2016-02-27:
@gfwilliams  | @tve  need some new ideas to get this right

MaBecker, 2016-03-15:
with all the optimization of const strings coming with the standard code we can have 1200 vars and 5200 heap ! 

DISABLE_LTO=1 USE_GRAPHICS=1 make 

````
>require("ESP8266").getState();
={
  "sdkVersion": "1.5.0",
  "cpuFrequency": 160, "freeHeap": 5200, "maxCon": 10,
  "flashMap": "4MB:512/512",
  "flashKB": 4096,
  "flashChip": "0xe0 0x4016"
 }
>process.memory();
={ "free": 1163, "usage": 37, "total": 1200, "history": 14 }
> 
````

````
Graphics.prototype.print = function() { 
  for (var y=0;y<this.getHeight();y++)
    console.log(new Uint8Array(this.buffer,this.getWidth()*y,this.getWidth()));
};

var g = Graphics.createArrayBuffer(8,8,8);
g.setColor(1);
g.drawString("X",0,0);
g.print();

/*
console output:
new Uint8Array([1, 0, 1, 0, 0, 0, 0, 0])
new Uint8Array([1, 0, 1, 0, 0, 0, 0, 0])
new Uint8Array([0, 1, 0, 0, 0, 0, 0, 0])
new Uint8Array([1, 0, 1, 0, 0, 0, 0, 0])
new Uint8Array([1, 0, 1, 0, 0, 0, 0, 0])
new Uint8Array(8)
new Uint8Array(8)
new Uint8Array(8)
*/
````

MaBecker, 2016-03-16:
this is this the heap size with bitmap_font_4x6 in flash

````
>require("ESP8266").getState();
={
  "sdkVersion": "1.5.0",
  "cpuFrequency": 160, "freeHeap": 5920, "maxCon": 10,
  "flashMap": "4MB:512/512",
  "flashKB": 4096,
  "flashChip": "0xe0 0x4016"
 }
````

wilberforce, 2016-03-16:
@MaBecker 

Does the heap size just end up with whatever ram is left, or is there a setting for this?

MaBecker, 2016-03-16:
@wilberforce  there is no setting for heap.

this is my understanding of how heap is influenced:

- increase vars and/or load more modules will decrease heap

- moving const strings and arrays to flash (".irom.literal") will increase heap

tve, 2016-03-16:
heap size is what's left

wilberforce, 2016-03-16:
How much heap is required? I understand this is application dependent. We start with 80K and I think we loose 1/2 to the interpreter - I'm wondering how the jsVars and heap available compare to ARM implementations

MaBecker, 2016-03-16:
try to write own lib to understand how .irom.literal stores data.

printf is not working they way I use it, any hints ?

````
extern int os_printf_plus(const char *format, ...)  __attribute__((format(printf, 1, 2)));
#define printf os_printf_plus

#include "jswrap_mabe.h" 
#include "jsinteractive.h" 

#define ESP_READ_FLASH_UINT8(ptr) ({ uint32_t __p = (uint32_t)(char*)(ptr); if (__p < 0x400000) __p|=0x40200000; volatile uint32_t __d = *(uint32_t*)(__p & (uint32_t)~3); ((uint8_t*)&__d)[__p & 3]; })
#define ESP_READ_FLASH_UINT16(ptr) ((ESP_READ_FLASH_UINT8(ptr)<<8) | ESP_READ_FLASH_UINT8(ptr+1) )    
//#define ESP_READ_FLASH_UINT16(ptr) ((ESP_READ_FLASH_UINT8(ptr+1)<<8) | ESP_READ_FLASH_UINT8(ptr) ) 
#define IN_FLASH_MEMORY   __attribute__ ((section(".irom.literal"))) //__attribute__((aligned(4))) 

unsigned short MYARRAY1[] = { 65,66,67,68,69,70,71,72 };
const unsigned short MYARRAY2[] = { 73,74,75,76,77,78,79,80 };
unsigned short MYARRAY3[] IN_FLASH_MEMORY = { 81,82,83,84,85,86,87,88 };

/*JSON{
  "type" : "class",
  "class" : "Mabe"
}*/

/*JSON{
  "type" : "staticmethod",
  "class" : "Mabe",
  "name" : "test01",
  "generate" : "jswrap_mabe_test01"
}*/

void jswrap_mabe_test01() {

	jsiConsolePrint("Hello MaBe!\r\n");

	printf("MYARRAY1: %u,%u\n",MYARRAY1[0],MYARRAY1[1]);
	printf("MYARRAY2: %u,%u\n",MYARRAY2[0],MYARRAY2[1]);

	printf("MYARRAY3: %u,%u\n",
		ESP_READ_FLASH_UINT16(&MYARRAY3[0]),
		ESP_READ_FLASH_UINT16(&MYARRAY3[1])
	);
}
/*
output
>Mabe.test01();
Hello MaBe!
=undefined
> 

xtensa-lx106-elf-objdump  -s libs/mabe/jswrap_mabe.o

Contents of section .data:
 0000 41004200 43004400 45004600 47004800  A.B.C.D.E.F.G.H.
Contents of section .rodata.str1.1:
 0000 4d594152 52415931 3a202575 2c25750a  MYARRAY1: %u,%u.
 0010 004d5941 52524159 323a2025 752c2575  .MYARRAY2: %u,%u
 0020 0a004d59 41525241 59333a20 25752c25  ..MYARRAY3: %u,%
 0030 750a00                               u..             
Contents of section .irom.literal:
 0000 48656c6c 6f204d61 4265210d 0a005100  Hello MaBe!...Q.
 0010 52005300 54005500 56005700 5800      R.S.T.U.V.W.X.  
Contents of section .rodata:
 0000 49004a00 4b004c00 4d004e00 4f005000  I.J.K.L.M.N.O.P.

*/
````

MaBecker, 2016-03-16:
using jsiConsolePrintf() with %d and ESP_READ_FLASH_UINT8 works 

````
	jsiConsolePrintf("MYARRAY1: %d,%d\r\n",MYARRAY1[0],MYARRAY1[1]);
	jsiConsolePrintf("MYARRAY2: %d,%d\n",MYARRAY2[0],MYARRAY2[1]);

	jsiConsolePrintf("MYARRAY3: %d,%d\n",
		ESP_READ_FLASH_UINT8(&MYARRAY3[0]),
		ESP_READ_FLASH_UINT8(&MYARRAY3[1])
	);

/*
>Mabe.test01();
Hello MaBe!
MYARRAY1: 65,66
MYARRAY2: 73,74
MYARRAY3: 81,82
=undefined
> 
*/
````

tve, 2016-03-17:
> How much heap is required?
YMMV :-)
I don't have much luck with 4KB or less. 5KB seems "OK" with a single outgoing mqtt connection and an incoming telnet connection. But you better try for yourself and include tests where the connection is slow/bad causing packet retransmissions. I still see the SDK allocating >2KB buffers at times, so if there's less free heap than that you're effectively toast.

[17 more replies]

closed 2016-04-05 by gfwilliams