espruino/Espruino #773: ESP8266 'save()' issue

olliephillips, 2015-12-29:
Build: 1v84.tve_master_c8c854e

Originally reported in Gitter, there is an error which occurs on ```save()```. I have the above build.

When connected over serial, the error message gets chops so there is not much to help, but when connected via IDE and wifi, concole gives me the following error information.

```
Writing..................
ERROR: Too big to save to flash (15805 vs 12284 bytes)
```

I had an earlier build to hand (1v84.tve_master_9bf51d3) on another ESP8266 and this issue does not occur with the same program/sketch. 

```
Erasing Flash.....
Writing............
Compressed 12276 bytes to 9774
```

gfwilliams, 2016-01-04:
@tve looks like the amount of flash pages dedicated to saving needs to be increased now that the RAM dedicated to variables has gone up. Probablye just something in `BOARD.py`?

I guess moving to 16 byte vars from 12 has meant that some code takes more space. Currently it's only RLE compression, which isn't ideal.

tve, 2016-01-04:
yup, which means moving some other stuff around and reducing the space available for code.

gfwilliams, 2016-01-04:
There is the LZ4_COMPRESS branch, which will do a much better job of compression - it does however use more flash for the compression algorithm. I was really hoping to find a very small compressor, but I haven't seen anything... Everyone seems more obsessed with speed than code size :)

tve, 2016-01-04:
Have you looked at heatshrink? http://spin.atomicobject.com/2013/03/14/heatshrink-embedded-data-compression/

gfwilliams, 2016-01-05:
Yes, that looks good... I'd seen that before, I'm not sure why I didn't use it in the branch :(

gfwilliams, 2016-01-05:
Ok, just added it (under LZ4_COMPRESS). It adds 1.5kB to the flash on ARM compiles. Do you reckon it's worth including it by default?

tve, 2016-01-05:
sounds like it pays for itself! :-)

gfwilliams, 2016-01-05:
Ok, just added - `SAVE_ON_FLASH` disables it though, as if you have only a small amount of RAM, the extra 1.5kB flash used to save 500 bytes isn't worth it :)

Also it has a neat hack now that will go back, delete the command history and try again if it runs out of space in Flash.

closed 2016-01-05 by gfwilliams

olliephillips, 2016-01-05:
Hi Guys, thanks for attention on this issue, I'm reading this as circa 30% better compression, so code in my original post (15805) would fit in the 12284 space with no tweaking to flash pages or impact on code space as mentioned, when @tve merges from upsteam ? Or is there more to it?