espruino/Espruino #774: Wifi.connect callback doesn't get the details 'connected' event gets

AkosLukacs, 2016-01-02:
An inconvenience at most, but maybe it would be useful to pass the connection details to the callback.
For example the following code, and it's output:
```javascript
var wifiLib = require("Wifi");

wifiLib.on('connected', function(){console.log('on connected', arguments);});

wifiLib.connect("p", {password: "****"}, function(err){
             console.log("connected callback", arguments);
});
```
Output:
```
connected callback [ null ]
on connected [
  {
    "ip": "192.168.1.6",
    "mask": "255.255.255.0",
    "gw": "192.168.1.1"
   }
 ]
```

maze1980, 2017-04-29:
The connect callback-functions returns an error string, if you supply a wrong password you'll see 'bad password'. I store the WiFi settings once and forever, and if the AP isn't available there's nothing the ESP8266 can do. Changing the callback arguments could break existing implementations ('err' and 'arguments' are both 'bad password' in case of an error or otherwise 'null').
I don't want to open another issue, just to mention this: getIP() returns the MAC address, too. The on.connected-callback function doesn't, Having said this, just call getIP() if you need connection details in the connect-callback function.

On the console:
```
require("Wifi").connect("ssid", {password: "password"})
require("Wifi").getIP()
require("Wifi").save()
require("Wifi").getDetails()
```
In the saved program:
```
var wifi = require("Wifi");
wifi.connect(
  wifi.getDetails().ssid,
  {password: wifi.getDetails().password},
  function(err) {
          if (err) {
            return;
          } else {
            console.log('omg it works');
          }
});
```

MaBecker, 2017-07-23:
@AkosLukacs or @maze1980  

Yes details for http://www.espruino.com/Reference#l_Wifi_connected differ from Wifi.geIP()
because mac is fixed.

So lets close this.

closed 2017-07-23 by MaBecker