espruino/Espruino #775: ESP8266 SPI doesn't work.

SpenceKonde, 2016-01-04:
Using an AT25512 EEPROM on ESP8266: 

```

var fakespi=new SPI();
pinMode(13,"output");
fakespi.setup({sck:D4,miso:D5,mosi:D12});
var eeprom=require("AT25").connect(fakespi, 128, 512, D13);

```

returns:

eeprom.read(0,10);
=new Uint8Array(10)

```

pinMode(13,"output");
SPI1.setup({sck:D4,miso:D5,mosi:D12});
var eeprom=require("AT25").connect(SPI1, 128, 512, D13);
doesnt:
eeprom.read(0,10);
=new Uint8Array([1, 1, 1, 1, 1, 1, 1, 1, 1, 1])

```

Trying to write does not result in data being written to the EEPROM. 

Should be returning (both return this from a pico), and I can write it from the pico too. 
[115, 99, 114, 101, 119, 98, 97, 108, 108, 101]

When using software SPI (on ESP8266), trying to send an array of bytes resets the board:
fakespi.send([0,0,0,0,255,255,128,64,255,64,192,128,255,255])

olliephillips, 2016-01-04:
Wild guess - I set pinMode with the alpha prefix you are just calling the number here. What happens if you add the letter prefix?

SpenceKonde, 2016-01-05:
No change in behavior.

commit d1bc108c (2016-01-18): fix/improve esp8266 SPI and I2C; fixes #775
closed 2016-01-18 by gfwilliams