espruino/Espruino #776: RTCPrescaler for LSI should be 30000 for 30kHz LSI on Pico micro controller (STM32F401CD)

dmcnaugh, 2016-01-04:
In Espruino/targets/stm32/jshardware.c
https://github.com/espruino/Espruino/blob/master/targets/stm32/jshardware.c
in the following code (lines: 667-680) the RTCPrescaler is set to 32768 assuming 32kHz for LSI on the Pico micro controller (STM32F401CD)

```C++
#ifdef USE_RTC
void jshSetupRTCPrescaler(bool isUsingLSI) {
  if (isUsingLSI) {
#ifdef STM32F1
    jshRTCPrescaler = 40000; // 40kHz for LSI on F1 parts
#else
    jshRTCPrescaler = 32768; // 32kHz for LSI
#endif
  } else {
    jshRTCPrescaler = 32768; // 32kHz for LSE
  }
  jshRTCPrescalerReciprocal = (unsigned short)((((unsigned int)JSSYSTIME_SECOND) << RTC_PRESCALER_RECIPROCAL_SHIFT) /  jshRTCPrescaler);
}
#endif
```

On the 7th line 

```C++
    jshRTCPrescaler = 32768; // 32kHz for LSI
```

According to the reference manual for the STM32F401 http://www.st.com/web/en/catalog/mmc/SC1169/SS1577/LN1810
the LSI frequency is 30kHz, requiring the RTCPrescaler to be set to 30000 as follows:

```C++
    jshRTCPrescaler = 30000; // 30kHz for LSI
```

What led me to look into this is noticing that the time on my Pico looses approximately 1 second every minute (without an external 32.768 kHz crystal attached). If you do the maths, this slip corresponds approximately to the delta in the prescaler.

Any chance this change can be made in the next release of the firmware, assuming it doesn't cause any other negative side effects?

gfwilliams, 2016-01-05:
Thanks, but where did you find out about the LSI frequency? Looking at page 81 of the datasheet [here](http://www.st.com/web/en/resource/technical/document/datasheet/DM00102166.pdf) it says:

| Symbol Parameter | Min | Typ | Max | Unit |
|---|---|---|---|---|
|fLSI|Frequency| 17| 32| 47| kHz|

So that's what I based it on (although based in this it should maybe be 32000). One problem is that the internal RTC on these is really variable (I didn't realise how bad initially) - on some it's better, but on some it's the full 10% off.

Is there a different figure somewhere else?

I'd actually planned on using the full speed oscillator while the chip is awake, and then swapping to LSI only when it's in deep sleep mode (maybe with some adjustment based on the measured LSI frequency relative to the HSE) - I haven't got a version that I was happy to pull into the main release yet though...

gfwilliams, 2016-01-05:
Having said all that, if 30k is nearer the average then it makes more sense to use that for now :)

dmcnaugh, 2016-01-05:
I apologise, you are right, the LSI frequency is 32 kHz requiring an RTCPrescaler value of 32000 rather than 32768.
I need to double check my facts before I write posts like this again.

gfwilliams, 2016-01-05:
Hey, no problem - at least we figured out that it needed changing :)

But for some reason the LSI is really poor on these, On the STM32F1 on the original boards it's really good, and so I just assumed that on a more modern chip it'd be the same or better.

Interestingly:

```
1 sec a minute = 1/60 = 1.6%
32768 / 32000 = 2.4%
```

So it's in the right ballpark. I imagine this would be a real improvement.

gfwilliams, 2016-01-05:
Just to add, an easy test is:

Short A0 and A1 (or any 2 pins, 1 of which does PWM)

```
analogWrite(A0, 0.5, {freq:1});
var last = getTime();
setWatch(function(e) {
 console.log(e.time - last);
 last = e.time;
}, A1, {repeat: true, edge:"rising"});
```

It should read near-ish 1 - it's comparing the LSI against the high speed oscillator, which is accurate to around 20ppm I think.

dmcnaugh, 2016-01-05:
Great, I'll try it out.
Meanwhile, do you have a script that dumps out the contents of the micro controller memory map formatted along the lines of the reference manual or would that be a useful tool to develop?

Also, I'm interested in your workflow with an issue like this from here. Does the issue get labeled in GitHub and closed with a release milestone, or do you have a seperate system?

gfwilliams, 2016-01-05:
I'm not quite sure what you mean about the memory map? You mean for accessing the peripherals with peek/poke?

Generally issues get closed when I commit a fix to GitHub (there are builds at http://www.espruino.com/binaries/git/commits/master/), but otherwise you only know what's in a release build by looking at the announcements or http://www.espruino.com/ChangeLog

closed 2016-01-05 by gfwilliams

gfwilliams, 2016-01-05:
Ok, I just did some tests on a few Picos I had hanging around, and the frequency of them was:

```
30100
34787
32956
31791
31804
30969
30562
34879
31431
```

So it's all over the place. The change may help a little, but probably not that much.

We really need something like is in this bug: https://github.com/espruino/Espruino/issues/240

Or potentially this branch, which I didn't think was stable enough to pull in: https://github.com/espruino/Espruino/tree/pico_systick_timer

Rather than calibrating the RTC, the branch just uses the SysTick timer, but when in deep sleep uses the RTC just to schedule a wakeup - it does mean that the current time won't survive a reset though.

dmcnaugh, 2016-01-05:
Ouch, that is disappointing. Makes me wish all the more that I'd soldered a 32.768kHz watch crystal onto my Pico before I'd attached an ESP8266 with the shim. Now there is no chance of getting to the pads on the board.

Well I'll try out the change when I see 1v85 released and see if it's better on my board.
Thanks.

gfwilliams, 2016-01-05:
You should actually be able to try it now: http://www.espruino.com/binaries/git/commits/2f9e0124fad90ebcf32746a02ff3fd56e24a35f3/

But yeah, it's just not really usable without some kind of calibration. A real shame as I'd intended to put an LSE on the board, but had to ditch it at the last minute (turns out the F4 needs a special low capacitance crystal).

commit 2f9e0124 (2016-01-05): Fix STM32F4 LSI clock frequency - should make the Pico's RTC a lot more accurate (fix #776)