espruino/Espruino #778: esp8266 reset

HyGy, 2016-01-05:
@tve here is the latest bug what I found.

version:
```javascript
 _____                 _
|   __|___ ___ ___ _ _|_|___ ___
|   __|_ -| . |  _| | | |   | . |
|_____|___|  _|_| |___|_|_|_|___|
          |_| http://espruino.com
 1v84.tve_master_d8dec28 Copyright 2015 G.Williams
WARNING: the esp8266 port is in beta!
Flash map 4MB:512/512, manuf 0xe0 chip 0x4016
```

code:

```javascript
require("ESP8266").logDebug(true);

var Wifi=require("Wifi");

var avaiableAccessPoints=[
  {
    ssid: 'a',
    pass: 'b'
  },
  {
    ssid: 'c',
    pass: 'd'
  }
];

var selectedAPIdx=null;

var connectToWifi = function() {

  console.log('connecting to wifi');

  Wifi.scan(
    function(apList) { 
      console.log(apList.length);
      for (var cikl=0; cikl<apList.length; cikl++)
      {
        console.log(apList[cikl]);
        for (var cikl1=0; cikl1<avaiableAccessPoints.length; cikl++)
        {
          if (avaiableAccessPoints[cikl1].ssid==apList[cikl])
          {
            selectedAPIdx=cikl1;

//            startConnect();

            return;
          }
        }
      }
    }
  );

};

E.on('init', function() {
  connectToWifi();
});

```

error log:

```javascript

   290> *** Espruino esp8266 booting
   290> 
   291> Restart info:
   291>   reason:   6=reset pin
   291>   exccause: 0
   292>   epc1:     0
   294>   epc2:     0
   296>   epc3:     0
   298>   excvaddr: 0
   300>   depc:     0
   302> Heap: 8568
   303> Variables: 1400 @16ea = 22400bytes
   307> Time sys=307640 rtc=57667
   310> Wifi.restore
   312> Wifi.restore: len=65535 vers=65535 crc=0xffffffff
   318> Wifi.restore cannot restore: version read=65535 exp=24, crc read=0xffffffff cacl=0x207eaf8d
   326> < Wifi init1, phy=3 mode=2
   329> mode : softAP(1a:fe:34:fa:3f:80)
   333> add if1
   334> dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
   342> bcn 100
   343> > initDone
   344> > jshInit
   346> RTC read: 1452079112 3306005 0x5d33fe1f (0xdeadbeef)
   351> RTC: cannot restore time
   354> < jshInit
   363> mode : sta(18:fe:34:fa:3f:80) + softAP(1a:fe:34:fa:3f:80)
   363> add if0
   363> f r0, Wifi.scan starting: mode=sta+ap
   367> < initDone
   369> tnSrv: creating...
   371> net_esp8266: listening socket 1 on port 23
   376> tnSrv: started sock=1
  2490> scandone
  2491> Wifi.scan completed, found 2

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 1396, room 16 
tail 4
chksum 0x89
load 0x3ffe8000, len 776, room 4 
tail 4
chksum 0xe8
load 0x3ffe8308, len 540, room 4 
tail 8
chksum 0xc0
csum 0xc0

2nd boot version : 1.4(b1)
  SPI Speed      : 80MHz
  SPI Mode       : QIO
  SPI Flash Size & Map: 32Mbit(512KB+512KB)
jump to run user1 @ 1000

   276> 
   276> 
   276> *** Espruino esp8266 booting
   276> 
   277> Restart info:
   277>   reason:   3=soft wdt
   277>   exccause: 4
   277>   epc1:     40223cd6
   280>   epc2:     0
   282>   epc3:     0
   284>   excvaddr: 0
   286>   depc:     0
   288> Heap: 8568
   290> Variables: 1400 @16ea = 22400bytes
   294> Time sys=294068 rtc=883911
   297> Wifi.restore
   299> Wifi.restore: len=65535 vers=65535 crc=0xffffffff
   304> Wifi.restore cannot restore: version read=65535 exp=24, crc read=0xffffffff cacl=0x207eaf8d
   313> < Wifi init1, phy=3 mode=2
   316> mode : softAP(1a:fe:34:fa:3f:80)
   319> add if1
   321> dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
   328> bcn 100
   329> > initDone
   331> > jshInit
   332> RTC read: 1452079112 3306005 0x5d33fe1f (0xdeadbeef)
   339> RTC: restore sys=338227 rtc=890999
   342> RTC: restored time: 1452105846 (delta=4292552290 cal=6228us)
   348> < jshInit
   356> mode : sta(18:fe:34:fa:3f:80) + softAP(1a:fe:34:fa:3f:80)
   356> add if0
   357> f r0, Wifi.scan starting: mode=sta+ap
   361> < initDone
   363> tnSrv: creating...
   365> net_esp8266: listening socket 1 on port 23
   369> tnSrv: started sock=1
  2484> scandone
  2484> Wifi.scan completed, found 2

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 1396, room 16 
tail 4
chksum 0x89
load 0x3ffe8000, len 776, room 4 
tail 4
chksum 0xe8
load 0x3ffe8308, len 540, room 4 
tail 8
chksum 0xc0
csum 0xc0

2nd boot version : 1.4(b1)
  SPI Speed      : 80MHz
  SPI Mode       : QIO
  SPI Flash Size & Map: 32Mbit(512KB+512KB)
jump to run user1 @ 1000```

tve, 2016-01-06:
soft wdt could well be an infinite loop, I see that you have:

```
for (var cikl1=0; cikl1<avaiableAccessPoints.length; cikl++)
```
that last increment bit looks like it's using the wrong variable and would explain the infinite loop?

HyGy, 2016-01-06:
Its just an example.  I tought te esp does not allow to reset from js code, couse every other cases it gives an error message.

tve, 2016-01-06:
I don't know what the bug is then. You run an infinite loop and the watchdog timer kicks in to reset the system. Works as designed and intended.

HyGy, 2016-01-06:
Okay, then sorry. No bug. I have not seen a system which has been reset in case of an endless cycle. But if it is wokring good than no bug, and sorry.

closed 2016-01-06 by HyGy