espruino/Espruino #794: ssd1306 OLED module, can't set brightness

yerpj, 2016-01-21:
Hi guys, 

I used to play a little with some SSD1306 OLED display (128x64). Unfortunately, I did not find any way of dimming the brightness. I had a look at the Adafruit driver, it seems that they implemented such a function. [see here](https://github.com/adafruit/Adafruit_Python_SSD1306/blob/master/Adafruit_SSD1306/SSD1306.py#L219).
It would be appreciable if someone could try to update the module. I will not take the risk to try by myself, because I did not understand the way the [SSD1306 module](http://www.espruino.com/modules/SSD1306.js) initializes the display. 
Basically, as I understood from the Adafruit driver, the command should be  0x81 (SSD1306_SETCONTRAST in the example) , followed by the contrast value (range [0..255]).

yerpj, 2016-01-21:
Finally I tried by myself : [#216](https://github.com/espruino/EspruinoDocs/pull/216) It seems to work, but is really not a good way to go, as the brightness is set once when calling ssd1306.connect(). It may be better to export a dedicated function, but this causes problems, as the display object is not meant to be accessed directly (rather accessed through the graphical object )

gfwilliams, 2016-01-22:
Thanks - I'll take a look at this. For what you're doing it'd be much easier if you modified the `update` function - it would then work on both I2C and SPI displays. 

However, I'll see about adding a setContrast function - it should be quite easy to add it like `flip` is handled at the moment.

gfwilliams, 2016-01-22:
Closing this, fixed in https://github.com/espruino/EspruinoDocs/commit/1556f307bf93c80bcc03ff2f9e5609d8936ec82b

closed 2016-01-22 by gfwilliams