espruino/Espruino #795: ST7565 for PI13264

MrTimcakes, 2016-01-25:
As mentioned in comment thread [271285](http://forum.espruino.com/conversations/271285/) incorporate change to allow for width changes. Change init to
```javascript
var g = require("ST7565").connect(spi, {rs:A7, cs:A5, rst:A6, width:132}, function() {
  g.clear();
});
```

Also possible to add PI13264 to docs with g.setContrast(0.5,6); specified as needed

gfwilliams, 2016-01-25:
What needs changing to make this work? I don't have a 132 pixel wide LCD here so I'm not sure if it's just as easy as changing the size of the Graphics object [here](https://github.com/espruino/EspruinoDocs/blob/master/devices/ST7565.js#L20).

MrTimcakes, 2016-01-25:
Yeah that should do it, currently that last 4 pixels don't get written. Thanks.

gfwilliams, 2016-01-25:
Any chance you could test? I don't want to do it and then have top back it out.

MrTimcakes, 2016-01-25:
Alright, will once I'm home again.

MrTimcakes, 2016-01-25:
I changed spi.write(new Uint8Array(this.buffer, 128*y, 128)); to spi.write(new Uint8Array(this.buffer, 132*y, 132)); as well and that worked fine.

MrTimcakes, 2016-01-25:
What I tested with:
```javascript
var exports={};

exports.connect = function(spi, RS, CS, RST, callback) {
  var g = Graphics.createArrayBuffer(132,64,1,{vertical_byte:true});

  CS.reset();
  if (RST!==undefined) RST.write(0);
  setTimeout(function() {
    if (RST!==undefined) RST.write(1);
    RS.reset(); // cmd
    spi.write([
     0xA3, // bias 7 (0xA2 for bias 9)
     0xA0, // no horiz flip
     0xC8, // vertical flip
     0x40, // top left start
     0x2F, // all power on (adafruit does this one at a time, but it doesn't seem to be required)
     0x20|2, // resistor divider contrast 2 (0..7)
     0xAF, // display on
     0x81, // dynamic contrast
     31 // 0..63
    ]);
    CS.set();
    if (callback) callback(g);
  }, 100);

  /* set contrast to a value between 0 and 1. 
  If div is specified, it should be an integer between 0 and 7 (2 is default) */
  g.setContrast = function(c, div) {
    if (c<0) c=0;
    if (c>1) c=1;
    CS.reset();
    RS.reset(); // cmd
    spi.write([0x81, 0|(c*63)]);
    if (div!==undefined) spi.write(0x20|div);
    CS.set();
  };

  /* Write to the screen */
  g.flip = function () {  
    CS.reset();
    for (var y=0;y<8;y++) {      
      RS.reset(); // cmd
      spi.write([0xB0|y/* page */,0x00/* col lower*/,0x10/* col upper*/]); 
      RS.set(); // data
      spi.write(new Uint8Array(this.buffer, 132*y, 132));
    }
    CS.set();
  };
  return g;
};

function draw() {
  g.clear();
  
  g.drawRect(0,0,131,63);
  
  g.flip();
}

var spi = new SPI();
spi.setup({ sck: B13, mosi: B15 });
var g = exports.connect(spi, C7 /* RS / DC */, C5 /* CS / CE */, C6 /* RST */, function() {
  g.setContrast(0.5,6);
  setInterval(draw, 1000);
});

gfwilliams, 2016-01-25:
Great - thanks! I'll get on this tomorrow.

MrTimcakes, 2016-01-25:
Awesome, Thanks. Your support is second to none.

closed 2016-01-26 by gfwilliams