espruino/Espruino #796: ESP8266 analogRead() returns 0~1023 integer not 0~1 float

SpenceKonde, 2016-01-26:
```
 _____                 _
|   __|___ ___ ___ _ _|_|___ ___
|   __|_ -| . |  _| | | |   | . |
|_____|___|  _|_| |___|_|_|_|___|
          |_| http://espruino.com
 1v84.tve_master_f35ac96 Copyright 2016 G.Williams
WARNING: the esp8266 port is in beta!
Flash map 4MB:512/512, manuf 0xe0 chip 0x4016
>echo(0);
=undefined
>analogRead(0);
=635

```

gfwilliams, 2016-01-26:
Should be easy enough to fix :) It should probably warn if you use the wrong pin too.

Also `jshPinAnalogFast` needs to shift left by 6, to use a full 16 bits rather than just 10 - it's used when using the Waveform class for recording audio/etc.

SpenceKonde, 2016-01-26:
There's only one analog pin on the ESP8266. As far as I can tell, analogRead() ignores the pin it's sent, and reads the ADC pin. Which seems like reasonable behavior, since that's the only pin it is valid for.

gfwilliams, 2016-01-26:
Well yes, except if you come from an Espruino board and you're expecting to be be able to use analog on any ADC-capable pin, you're going to get confused that you get a value, but not from the pin you're expecting.

gfwilliams, 2016-02-18:
Fixed in https://github.com/espruino/Espruino/commit/cb0232b70e054133c3e3babe19dc675529b5a372

gfwilliams, 2016-02-18:
Oh, and I can't add A0, because it'll shift all the pin numbers up, and that'll screw everyone up who was using `Pin()` :(

closed 2016-02-18 by gfwilliams