espruino/Espruino #797: ESP8266 - console.log() while processing http request crashes or resets board

SpenceKonde, 2016-01-26:
Attempting to print to console within http server callback doesn't work...

```

function onPageRequest(req, res) {
  var a = url.parse(req.url, true);
  console.log("test");
  res.writeHead(200);
  res.end("Hello World"); 
}
require("http").createServer(onPageRequest).listen(80);

```

Requesting a page will crash and reset the board, or put it into a state where it needs to be manually reset to work. Using 

```

 _____                 _
|   __|___ ___ ___ _ _|_|___ ___
|   __|_ -| . |  _| | | |   | . |
|_____|___|  _|_| |___|_|_|_|___|
          |_| http://espruino.com
 1v84.tve_master_f35ac96 Copyright 2016 G.Williams
WARNING: the esp8266 port is in beta!
Flash map 4MB:512/512, manuf 0xe0 chip 0x4016

```

wilberforce, 2016-02-11:
It works fine on this build, with a nodemcu board:
````
 _____                 _
|   __|___ ___ ___ _ _|_|___ ___
|   __|_ -| . |  _| | | |   | . |
|_____|___|  _|_| |___|_|_|_|___|
          |_| http://espruino.com
 1v84.255 Copyright 2016 G.Williams
WARNING: the esp8266 port is in beta!
Flash map 4MB:512/512, manuf 0xe0 chip 0x4016
>function onPageRequest(req, res) {
:  var a = url.parse(req.url, true);
:  console.log("test");
:  res.writeHead(200);
:  res.end("Hello World");
:}
=function (req,res) { ... }
>require("http").createServer(onPageRequest).listen(80);
=undefined
test
test
````

Have you connected to a wifi station and used the save command, so that the network comes up before trying the above code?

I use:
````
var wifi=require("Wifi");
function join(ssid,password) {
  print('joining');
	wifi.connect(ssid,
                 {password: password}, 
                 function (err, ipInfo) {
      ipInfo = wifi.getIP();
      print("Wifi IP info: " + JSON.stringify(ipInfo));
      wifi.save();

	});
}

join("mysid","mypassword"); // change these...
````

I have found that to update firmware, it is best to erase the flash first:
..\..\esptool-master\esptool.py --port COM7 --baud 115200 erase_flash

and burned with:
..\..\esptool-master\esptool.py --port COM7 --baud 115200 write_flash --flash_freq 80m --flash_mode dio --flash_size 32m 0x0000 "boot_v1.4(b1).bin" 0x1000 espruino_esp8266_user1.bin 0x37E000 blank.bin

These are setting for windows with my paths...

I can also get it working with the same build:

 _____                 _
|   __|___ ___ ___ _ _|_|___ ___
|   __|_ -| . |  _| | | |   | . |
|_____|___|  _|_| |___|_|_|_|___|
          |_| http://espruino.com
 1v84.tve_master_f35ac96 Copyright 2016 G.Williams
WARNING: the esp8266 port is in beta!
Flash map 4MB:512/512, manuf 0xe0 chip 0x4016
>function onPageRequest(req, res) {
:  var a = url.parse(req.url, true);
:  console.log("test");
:  res.writeHead(200);
:  res.end("Hello World");
:}
=function (req,res) { ... }
>require("http").createServer(onPageRequest).listen(80);
=undefined
test
test
>

MaBecker, 2016-12-10:
@SpenceKonde can this be closed ?

closed 2017-07-21 by MaBecker