espruino/Espruino #812: telnet console switches back to default console on close

tve, 2016-02-27:
When a telnet session is started the console is moved from the current console device to telnet. When the telnet connection is closed the console is moved back to the default console device instead of the previous console device.

gfwilliams, 2016-02-29:
It'd be nice to have a general way of handling this, since it's the same issue as USB... I wonder whether `connected` and `disconnected` events on `E` could be fired (and if there was no handler we'd go for the default behaviour).

... Or maybe `setConsole` could take a boolean argument that `forced` the console, so it would stay regardless of what happened.

tve, 2016-02-29:
I was thinking about adding a setDefaultConsole() call... Would need to store the setting in flash?

tve, 2016-02-29:
oops closed by mistake

gfwilliams, 2016-02-29:
Yes, that could be a good idea I guess. The simplest is to store it as a variable and then generate the JS code to set it [here(ish)](https://github.com/espruino/Espruino/blob/master/src/jsinteractive.c#L626) - then it gets `dumped` correctly, but also re-initialised when saved.

On the Espruino boards there are 2 main use-cases:

* Stop auto-switching to USB console, so USB can be used for something else (not directly relevant to Telnet)
* Stop going to Serial1 when USB disconnected

So `setDefaultConsole` would help with the second, but not the first

commit 56099e1e (2016-04-26): fix #812: make telnet switch back to prior console device
closed 2016-04-26 by gfwilliams