espruino/Espruino #815: Multiplatform equivalent to ESP8266.getFreeFlash()

gfwilliams, 2016-03-04:
In Espruino boards there's some info in `process.memory()`, but it doesn't cover the case where there are several non-continuous chunks of memory.

We ought to add something global (maybe via some API in `jshardware.h`) that modules like `FlashEEPROM` can use.

wilberforce, 2016-03-05:
Perhaps process.memory().flash structure , that would be an array of address, and length like the current esp8266 function?

wilberforce, 2016-03-07:
Correction - the flash module can only read the first 1Mb of memory.

> Perhaps process.memory().flash array, that would be an array of address, and length like the current esp8266 function? 

As well as address and length, the page size might be useful too.

gfwilliams, 2016-03-07:
> the page size might be useful too.

It already exists:

http://www.espruino.com/Reference#l_Flash_getPage

Perhaps this should actually just be a function in the `flash` library? It maybe makes more sense.

wilberforce, 2016-03-07:
Yes  I was aware of this to get the page size.
As the flash code is board dependant, it would be a great place to add this information.

commit b61d388f (2016-03-07): Add `require("Flash").getFree()` as multiplatform way to find free flash pages (fix #815)
closed 2016-03-07 by gfwilliams