espruino/Espruino #817: jspEvaluate can use MemoryArea rather than creating a string

gfwilliams, 2016-03-08:


wilberforce, 2016-03-30:
Please expand on what your are thinking here as I think this means we may be able to read modules from flash without copying into memory first

gfwilliams, 2016-03-30:
Right now `jspEvaluate` takes a `char*`, creates a new JS string from it and executes that. It could very easily execute without the copy into the JS string.

To be honest it's not high priority because `jspEvaluate` isn't used a great deal.

This has been covered many times before on the forum and GitHub, but sure - you can use 'Modules.addModule` with a MemoryArea, but *that won't magically store your module in flash*. The string is still executed and the result of execution is stored in RAM. **The end result after uploading is the exact same memory usage.**

With some huge hacks to Espruino you could maybe store the module *source* in flash with memoryarea, but it's not saving you that much as the function declarations still have to be in RAM, as does 1 JsVar representing the memoryarea itself. It means that unless your function has more than 16 characters of code in it you're going to be *worse off* using flash for it.

**By far the best solution** on systems with lots of flash is to use the [variable cache branch](https://github.com/espruino/Espruino/issues/62). This stores *everything* in flash, and only uses RAM as a cache for the most-used JS Objects.

I spent quite a bit of time on it, but it still needs some tweaking to get it passing all the tests, and to make a flash filesystem for the variables. Sadly nobody else has picked it up and I don't have the time given ESP8266 is the only thing that really benefits.

Probably the most useful solution for all involved is to make the Web IDE capable of loading all modules into one file and then minifying. I just added an issue for this here: https://github.com/espruino/EspruinoTools/issues/27

While it doesn't use flash, it would reduce code size so much that it's unlikely anyone would care.

commit 92b8b58c (2016-03-31): `jsvEvaluate` now uses memory area for execution of JS strings (fix #817)
            Add `E.setBootCode` to allow JS scripts to be run without being in RAM (fix #740)
closed 2016-03-31 by gfwilliams