espruino/Espruino #830: How to disable default serial port REPL

ceremcem, 2016-04-03:
Hi, 

I want to create a transparent serial <-> wifi bridge and in order  to make this happen, I need to disable the default REPL at some point of execution. 

Closest documentation I could find is "Console Over Serial" section of [this](http://www.espruino.com/USART) page but it doesn't help either. 

How can I disable the REPL console with an application code?

gfwilliams, 2016-04-04:
You can simply do `LoopbackA.setConsole()` - this'll put the console on Loopback - where you can access it by reading/writing `LoopbackB`.

For questions like this (as it's not a bug), please can you ask on the forums. I have already answered this (or similar) questions several times previously on there.

closed 2016-04-04 by gfwilliams

ceremcem, 2016-04-04:
Wow! This is clearly the best of good news today! My application works right now and it made me feel like in the last scene of [this](https://www.youtube.com/watch?v=4K39UWxdm0U). Thank you :) 

But, isn't it  still a bug as I found the issue in the documentation and documentation says it is not implemented right now (or am I constantly misunderstanding what I read): 

> The esp8266 has two UARTS. UART0 uses gpio1 for TX and gpio3 for RX and is used by the Espruino console. It could be reused for application purposes, but that is not currently implemented. UART1 uses gpio2 for TX and RX is not really usable due to being used for the SDIO flash chip. UART1 TX is used by the debug and could be reused for application purposes, but that is not currently implemented.

gfwilliams, 2016-04-04:
Yes, you're right. I'll add an issue in EspruinoDocs, where the documentation is held.