espruino/Espruino #832: Calling USB.setConsole() causes Reset on esp8266

ceremcem, 2016-04-03:
I'm using Espruino v1.85 and following application code causes the reset (see below) 15 seconds after the power on: 

```
E.on('init', function(){
  setTimeout(USB.setConsole, 15000);
  return connectToWifi();
});
```

```
ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x40100000, len 1396, room 16 
tail 4
chksum 0x89
load 0x3ffe8000, len 776, room 4 
tail 4
chksum 0xe8
load 0x3ffe8308, len 540, room 4 
tail 8
chksum 0xc0
csum 0xc0

2nd boot version : 1.4(b1)
  SPI Speed      : 40MHz
  SPI Mode       : QIO
  SPI Flash Size & Map: 4Mbit(256KB+256KB)
jump to run user1 @ 1000

rlrl���rLoading 2685 bytes from flash...
```

gfwilliams, 2016-04-04:
Thanks - will look at this.

It could be because of the way you're calling it - doing `a=USB.setConsole;a()` means that you are executing the function, but without setting `this`.

However, `USB` shouldn't even be defined for ESP8266.

MaBecker, 2016-12-20:
Strange, there is no device named in boards/ESP8266.py but USB is available ?!

```
>process.env.VERSION
="1v90.2079"
>process.env.GIT_COMMIT
="f39dd3b75668f059487d6cbe1ba8fa6463885668"

>USB
=Serial {  }
```

How can this be disabled for ESP boards ?

gfwilliams, 2016-12-20:
```
/*JSON{
  "type" : "object",
  "name" : "USB",
  "instanceof" : "Serial",
  "#ifdef" : "USB"
}
The USB Serial port
 */
```

So for some reason, `build_jswrapper.py` thinks USB is defined somewhere?

MaBecker, 2017-01-05:
@gfwilliams  any hints where to start searching ?

gfwilliams, 2017-01-05:
I'd look in `scripts/build_jswrapper.py` (it's probably common.py called from there) - if you check `gen/jswrapper.c` after the build you'll see `USB` in there I guess - `build_jswrapper.py` shouldn't be putting it in if `USB` isn't defined

MaBecker, 2017-01-06:
Thanks, did some  changes in `scripts/common.py` but without success :(

Then I checked a Puck.js build - same result, something in that gen chain is always adding USB ?!

o section in common.py 

https://github.com/espruino/Espruino/blob/master/scripts/common.py#L114-L115

https://github.com/espruino/Espruino/blob/master/scripts/common.py#L172-L186

o section in jswrap_serial.c

https://github.com/espruino/Espruino/blob/master/src/jswrap_serial.c#L81-L86 

shrinked call

`
python scripts/build_jswrapper.py src/jswrap_serial.c -DNO_ASSERT -DRELEASE  -DUSE_DEBUGGER -DUSE_TAB_COMPLETE -DUSE_HEATSHRINK -DUSE_MATH -DUSE_GRAPHICS -DUSE_NET -DUSE_ESP8266 -DUSE_TELNET -DUSE_CRYPTO -DUSE_OPTIMIZE_PRINTF -D__ETS__ -DICACHE_FLASH -DXTENSA -DUSE_US_TIMER -DEMBEDDED -DESP8266 -BESP8266_BOARD -F/esptools/Espruino/ESP8266/MaBecker/gen/jswrapper.c
`

output - expected USB to be droped ?!

`
BOARD ESP8266_BOARD
Script location /Volumes/ESPTools/Espruino/ESP8266/MaBecker/scripts
Got define from board: USE_NET
Got define from board: USE_TELNET
Got define from board: USE_GRAPHICS
Got define from board: USE_CRYPTO
Using files from command line
Got #DEFINES:
   USE_NET
   USE_TELNET
   USE_GRAPHICS
   USE_CRYPTO
   NO_ASSERT
   RELEASE
   USE_DEBUGGER
   USE_TAB_COMPLETE
   USE_HEATSHRINK
   USE_MATH
   USE_GRAPHICS
   USE_NET
   USE_ESP8266
   USE_TELNET
   USE_CRYPTO
   USE_OPTIMIZE_PRINTF
   __ETS__
   ICACHE_FLASH
   XTENSA
   USE_US_TIMER
   EMBEDDED
   ESP8266
   USART_COUNT=2
   SPI_COUNT=1
   I2C_COUNT=1
   defined(USB)=False
Scanning src/jswrap_serial.c
Dropped Serial3  because of #if USART_COUNT>=3 -> 2>=3
Dropped Serial4  because of #if USART_COUNT>=4 -> 2>=4
Dropped Serial5  because of #if USART_COUNT>=5 -> 2>=5
Dropped Serial6  because of #if USART_COUNT>=6 -> 2>=6
WARNING: error evaluating 'defined(USE_TELNET)' - from 'defined(USE_TELNET)'
Scanning finished.

`

gfwilliams, 2017-01-06:
Thanks - after all that it was just an error in the way the JSON was formed

closed 2017-01-06 by gfwilliams

MaBecker, 2017-01-06:
No, thank you.

commit bb52ec9d (2017-01-06): Fix bug that caused Bluetooth/TV/USB objects to be added in devices that didn't support them (fix #832)