espruino/Espruino #833: pinMode() not stored in flash on NUCLEO-F411RE?

yerpj, 2016-04-04:
Hello,

Working on NUCLEO-F411RE, I tried to save() some code into flash, including this line:
```pinMode(C7,'output');``` 
Then, dumping the code with dump() indicates me that the line of code magically transformed to ```digitalWrite(C7,1);``` 

Then I tried to save only this pinMode() instruction into flash. after another dump(), the only line remaining in flash is also ```digitalWrite(C7,1);``` 

This NUCLEO is running the 1v85 from the official archive.

gfwilliams, 2016-04-04:
That's expected - it'll happen with any STM32 board. It's because `pinMode(C7,'output');` has the same effect as `digitalWrite(C7,1);`, and the commands are reconstructed from the current state of the hardware.

If you want your code to be remembered character for character, put it in `onInit()`.

Is that what you were asking, or have I missed something?

yerpj, 2016-04-04:
I think you got it. But I don't understand how `pinMode(C7,'output')` could have the same effect as `digitalWrite(C7,1)` ? 
Direction is selected by the bit-config in GPIOx_MODER, according to the datasheet, and the pin value when in output is selected by the bit-config in GPIOx_ODR register. I think those two operations are completely different. Furthermore IMO, in some case, it could be unsafe to force `1`on a pin without the user agreement.

gfwilliams, 2016-04-04:
Behind the scenes, `digitalWrite(C7,1)` basically calls `pinMode(C7,'output')` first, and also sets the output state to 1. This isn't like Arduino but is well documented and has been done to be more beginner-friendly (if you have already done pinMode, it uses the mode you had previously).

Now I'm not sure why the pin output state has been set to 1. Have you uploaded other code that might also make it output a 1? When I do `pinMode(A5,'output')` on the Pico it translates to `digitalWrite(A5,0);`.

> Furthermore IMO, in some case, it could be unsafe to force 1 on a pin without the user agreement.

Well, it's got to force it to something if you do `pinMode(C7,'output')`. In some cases, `0` could be dangerous too. At least `digitalWrite` is explicit about what's happening :)

yerpj, 2016-04-04:
> Have you uploaded other code that might also make it output a 1?
No, the only code uploaded was `pinMode(C7,'output')` 

Yes that's true, forcing a `0` could be dangerous as well :-) 
But what seems odd to me, let's say, you upload the code 
`pinMode(C7,'output');
digitalRead(C7)`
Then , when reading the C7 pin, its state will stay to `1` because of the previous explicit call to pinMode(), right? 
But with 
`digitalWrite(C7,1);
digitalRead(C7)` 
reading the C7 pin will release it to `0`. It means that pinMode does not behave exactly as digitalWrite. 
I understand that things are done this way and I don't think another way would be better, but what if `digitalWrite()` could act as pinMode, like forcing the pin mode to `'output'`? This way, calling a `digitalRead()`after a `digitalWrite()`would simply give the current OUTPUT value instead as a unwanted INPUT value?

gfwilliams, 2016-04-04:
That's a really good point. Reopening...

So it should use `pinMode` when `pinMode` has been used to set the state - as if you use `digitalWrite` it doesn't 'force' the pin to the correct mode.

The `1` might actually be because you used `digitalWrite(C7,1)` previously and only did `reset()` to reset. While Espruino resets MODER on reset, I'm not sure it resets ODR - so that could potentially be another issue (although I think it's rare you'd hit that).

yerpj, 2016-04-04:
> The 1 might actually be because you used digitalWrite(C7,1) previously and only did reset() to reset. While Espruino resets MODER on reset, I'm not sure it resets ODR - so that could potentially be another issue (although I think it's rare you'd hit that).

That's seems correct, I only `reset()` the board, not a hard reset. I just tried with a hard reset, and the pin is now `0` after a `pinMode`.

In term of behaviour, I guess that an explicit `digitalWrite()` implies that the user wants the corresponding pin to behave as an output. This is why I would suggest that it might be a good improvement to force the direction inside the `digitalWrite()` function. That said, I don't know if doing a `digitalWrite()` on a pin which has been configured as input with `pinMode()` has a specific meaning? Sometimes it can be understood as a way of configuring the input pull up/down (msp430 are made this way. the GPIO register reflecting the output state is re-used as a pull-up/pull-down register when in input mode). I personally think this is not a good way to go.

gfwilliams, 2016-04-04:
It does set the mode, if you haven't already done so with `pinMode`. The reason is that you can set the mode to `output_opendrain`, and then you expect to be able to use `digitalWrite`/etc as normal.

yerpj, 2016-04-04:
I thought that `digitalWrite()` could set the pin direction to 'output' only if the latter was not yet configured with `pinMode()`. Anyway, if we would like to set up an output as open-drain, we actually have to call `pinMode(xx,'output_opendrain')`. By knowing if an explicit `pinMode()`was done previously, `digitalWrite()`could be able to decide wether it has to keep the mode as is or configure it by default to output push-pull. 

As you said before, another way of being sure of the pins config is to set them inside the `onInit()` function...

gfwilliams, 2016-04-05:
> I thought that digitalWrite() could set the pin direction to 'output' only if the latter was not yet configured with pinMode()

Yes - that's actually exactly what happens at the moment...

yerpj, 2016-04-05:
Then I don't understand why a successive `digitalWrite(pin,1);digitalRead(pin)` effectively resets `pin`to `0`.

gfwilliams, 2016-04-05:
Because `digitalRead()` also sets the pin direction to 'input' if the pin wasn't configured with `pinMode()`.

It makes a lot of sense for beginners. If you don't know about `pinMode` everything 'just works', and if you do, it also works as you expect.

yerpj, 2016-04-05:
I do not completely agree with this philosophy. In my point of view, when a user [even a beginner] calls `digitalWrite()` prior to `digitalRead()`, he clearly wants to set a logical level on the corresponding output pin, then he wants to read it back. Then if he really wants to change the pin behaviour, he will have to call explicitely `pinMode()` before reading the pin . 
Anyway, as said before, if you are comfortable with the current behaviour, I suggest that things remains as is.

gfwilliams, 2016-04-05:
Yeah, I think the main points that need fixing are:

* `reset()` doesn't reset ODR to power-on values
* When 'dumping' code, we should use `pinMode()` if we know the pin state had been set up manually

commit 6ca54db0 (2016-04-07): Ensure that pinMode/digitalWrite is re-constituted properly by dump() and save() (fix #833)
commit caddc7ab (2016-05-05): Ensure that pinMode/digitalWrite is re-constituted properly by dump() and save() (fix #833)
closed 2016-04-07 by gfwilliams