espruino/Espruino #834: setWatch is not firing a function sometimes

ceremcem, 2016-04-05:
I'm using v1.85 on STM32F407-Discovery and following code works as expected 99% of the time but fails somehow at somepoint and do not fire the `fireGunStart` function: 
```
setupDigitalIo = function(){
  setWatch(fireGunStart, BTN1, {
    repeat: true,
    edge: 'both',
    debounce: 30
  });
};
```

Full source code is here: 
```
var ser2, callAfter, continueFiring, fireGunStart, fireGun, setupDigitalIo, detachRepl, setupSerials, onInit;
ser2 = Serial2;
callAfter = function(delay, func){
  return setTimeout(func, delay);
};
continueFiring = true;
fireGunStart = function(pin){
  console.log("fire gun start: ", pin);
  if (pin.state) {
    continueFiring = true;
    fireGun();
  } else {
    continueFiring = false;
  }
};
fireGun = function(){
  ser2.write("nabernabernabernabernaber" + "\r\n");
  if (continueFiring) {
    callAfter(10, fireGun);
  }
};
setupDigitalIo = function(){
  setWatch(fireGunStart, BTN1, {
    repeat: true,
    edge: 'both',
    debounce: 30
  });
};
detachRepl = function(){
  console.log("Disabling REPL console!");
  LoopbackA.setConsole();
};
setupSerials = function(){
  var baud;
  baud = 115200;
  ser2.setup(baud);
};
onInit = function(){
  setupSerials();
  return setupDigitalIo();
};
```
How can I debug this issue?

gfwilliams, 2016-04-05:
It could be because of button bounce - what can happen is the IRQ gets called by the hardware when something changes, but it doesn't know what the state of the pin is so it has to check. Unfortunately the button can change state so quickly that it interrupts for the button release, but the button is in mid-bounce and when it reads the value it reads it as pressed. It means that even the debounce thinks the button is still pressed.

Sadly it's almost impossible to handle using interrupts unless you resort to polling.

The best solution is probably;

```
fireGunStart = function(pin){
  console.log("fire gun start: ", pin);
  fireGun();
};
fireGun = function(){
  ser2.write("nabernabernabernabernaber" + "\r\n");
  if (digitalRead(BTN)) {
    callAfter(10, fireGun);
  }
};
  setWatch(fireGunStart, BTN1, {
    repeat: true,
    edge: 'rising'
  });
```

But you could put a small capacitor across the button, which would reduce the bounce frequency and should also fix it,

closed 2016-04-05 by gfwilliams

ceremcem, 2016-04-05:
If I'm not wrong, if there is such a possibility then I have to use polling to detect the rising edge, thus `setWatch()` is totally out of usage. So the only option is using polling, which is a HIGHLY cpu intensive approach. 

I may not be able to add a capacitor in every situation, because this digital input might be set from a different source than a simple button (eg. encoder). 

My proposal is that if there is a limitation with espruino itself (it may choose to allocate a limited cpu time or something) then espruino should make a last check via polling once more so setWatch becomes trustworthy.