espruino/Espruino #837: FlashEEPROM.readMem() causes reset on esp8266

ceremcem, 2016-04-08:
While following [the documentation], I faced with the following situation: 
```
getFile = function(){
  var f;
  f = new (require("FlashEEPROM"))(0x076000);
  f.endAddr = f.addr + 1024;
  return f;
};
```

```
>f = getFile()
={ 
  "flash": function () { [native code] }, 
  "addr": 483328, "endAddr": 484352 }
>f.read(0)
=new Uint8Array([104, 101, 108, 108, 111])
>E.toString(f.read(0))
="hello"
>f.readMem(0)
=" ets Jan  8 2013,rst cause:4, boot mode:(3,7)
wdt reset
load 0x40100000, len 1396, room 16 
tail 4
chksum 0x89
load 0x3ffe8000, len 776, room 4 
tail 4
chksum 0xe8
load 0x3ffe8308, len 540, room 4 
tail 8
chksum 0xc0
csum 0xc0
2nd boot version : 1.4(b1)
  SPI Speed      : 40MHz
  SPI Mode       : QIO
  SPI Flash Size & Map: 4Mbit(256KB+256KB)
jump to run user1 @ 1000
rlrl��Loading 6525 bytes from flash...
Running onInit()...
connected? err= null info { 
  "ip": "192.168.2.193", 
  "netmask": "255.255.255.0", 
  "gw": "192.168.2.101", 
  "mac": "18:fe:34:d6:72:f6"
 }
```

gfwilliams, 2016-04-08:
Thanks - I guess this might be something to do with how flash memory is mapped into the address space. Presumably `E.memoryArea` should add some offset to addresses that are in flash.

For now, just use `f.read()` - which should work fine? `readMem` is only really useful when you're storing massive amounts of data.

ceremcem, 2016-04-08:
Okay, `f.read()` works fine and suffices right now.

MaBecker, 2016-12-09:
@ceremcem - can this be closed ?

gulfaraz, 2016-12-15:
@MaBecker readMem still resets the esp8266

MaBecker, 2016-12-17:
@gulfaraz  - can you add sample code  ?

ceremcem, 2016-12-17:
@MaBecker  Is my sample code passing your tests?

MaBecker, 2016-12-17:
@ceremcem - did some testing

 `E.memoryArea()`  is the reboot cause for this reboot on ESPs

MaBecker, 2016-12-20:
@gfwilliams  - I guess this has todo with "ESP reads 4byte from flash"  

https://github.com/espruino/Espruino/blob/master/src/jswrap_espruino.c#L751

Can you give some hints how to fix this ?

gfwilliams, 2016-12-20:
The ESP8266 is incapable of reading just a single byte from flash memory, it must read 4.

There is a bit of code somewhere - an exception handler - which will handle the exception and read the single byte.

I think @tve was against it at the time as it's pretty slow, but I think in this case it might be a good idea. Modifying all of Espruino so it never attempts to access a single byte element of a flat string at any point would be a bit of a nightmare (and would leave the code looking pretty ragged) - but the exception handler would fix the few cases where it was a problem.

MaBecker, 2017-07-24:
So its f.read() for now and lets close this.

closed 2017-07-24 by MaBecker