espruino/Espruino #840: Frozen prompt on ESP8266

pbuyle, 2016-04-13:
After flashing on a [NodeMCU devkit 0.9](https://github.com/nodemcu/nodemcu-devkit) (ESP-12 with 4Mb flash) with 
```
esptool.py --baud 460800 write_flash   --flash_freq 80m --flash_mode qio --flash_size 32m   0x0000 "boot_v1.4(b1).bin" 0x1000 espruino_esp8266_user1.bin 0x37E000 blank.bin
```

When connecting with the IDE, I got a frozen terminal in the left-hand panel with only
```
Connected
>
```

In the IDE log, I find this
```
[...]
Loaded code from storage.
>>> Connecting...
Set Slow Write = true
[object Object]
Connected [object Object]
Found a prompt... great!
>>> Sending...
---> "echo(0);\nconsole.log(\"<\",\"<<\",JSON.stringify(process.env),\">>\",\">\");echo(1);\n"
Sending block "echo(0);\nconsol", wait 50ms
Sent
Sending block "e.log(\"<\",\"<<\",", wait 50ms
Sent
Sending block "JSON.stringify(", wait 50ms
Sent
Sending block "process.env),\">", wait 50ms
Sent
Sending block ">\",\">\");echo(1)", wait 50ms
Sent
Sending block ";\n", wait 50ms
Sent
>>> Sent
WARNING: No result found - just got ""
Device found (connectionId=12)
>>> Connected to port /dev/ttyUSB0
Sending block "d", wait 50ms
Sent
Sending block "d", wait 50ms
Sent
Sending block "d", wait 50ms
Sent
Sending block "d", wait 50ms
Sent
Sending block "d", wait 50ms
Sent
Sending block "d", wait 50ms
Sent
```

Using `minicom -b 9600 -D /dev/ttyUSB0` produce the same result.

Note: my board look exactly like this  [image]

gfwilliams, 2016-04-13:
I'd be 99% sure you're just using the wroing baud rate - try 115200.

ESP8266 is the **only** board that runs Espruino that doesn't use 9600.

gfwilliams, 2016-04-13:
Closing this, as this was also asked on the forum: http://forum.espruino.com/conversations/279176/?offset=50#comment12918125

Which is the best place - this is unlikely to be a bug in Espruino as it works on so many other ESP8266 boards.

closed 2016-04-13 by gfwilliams